polarofficial/polar-ble-sdk #542: Is it necessary to call firstTimeUse on Polar 360 to get started

abbjetmus, 2025-01-23:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ X] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ x] Other
- [ ] Device is not relevant for this question

**Description:**

Hi! I have a question regarding Polar 360. Our company has bought some Polar 360 sensors from you guys. To communicate we use a Flutter wrapper on the native SDKs for Android and iOS. For that wrapper I have added offline support and we mainly store PPI data. The flow works fine for Polar Verity Sense. But for Polar 360, do I need to add support for firstTimeUse function call to get started with the sensor to record offline PPI? And I guess the sdk api is the same for Verity Sense and 360?

[https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/FirstTimeUse.md](https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/FirstTimeUse.md
)

rkangast, 2025-01-24:
Hi there,

Yes, the first time use is required to be done for Polar 360 before any offline (or online) measurement can be performed. So please do add support for that in your application. SDK works similarly as with Verity Sense otherwise.

abbjetmus, 2025-01-24:
Thank you for the answer @rkangast, I've added support for doFirstTimeUse and trying it on Android.
After scanning and connecting successfully I call doFirstTimeUse() without any errors so I expect it to succeed as well.

But trying to call function like getOfflineRecordingStatus or startOfflineRecording I get errors. These functions works well with Polar Verity Sense. The error message looks like:

```error
E/flutter (18923): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command GET_MEASUREMENT_STATUS error: ERROR_INVALID_STATE, pmd cp command GET_MEASUREMENT_STATUS error: ERROR_INVALID_STATE, null, null)
E/flutter (18923): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
E/flutter (18923): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
E/flutter (18923): <asynchronous suspension>
E/flutter (18923): #2      Polar.getOfflineRecordingStatus (package:polar/src/polar_base.dart:656:20)
E/flutter (18923): <asynchronous suspension>
E/flutter (18923): #3      _PolarListPageState.build.<anonymous closure>.<anonymous closure> (package:polar_variety_sense_example/pages/polar_list_page.dart:187:25)
E/flutter (18923): <asynchronous suspension>
E/flutter (18923): 
```

```error
E/flutter (18923): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(ERROR_STARTING_RECORDING, pmd cp command REQUEST_MEASUREMENT_START error: ERROR_INVALID_STATE, null, null)
E/flutter (18923): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
E/flutter (18923): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
E/flutter (18923): <asynchronous suspension>
E/flutter (18923): #2      Polar.startOfflineRecording (package:polar/src/polar_base.dart:617:5)
E/flutter (18923): <asynchronous suspension>
E/flutter (18923): #3      _PolarListPageState.build.<anonymous closure>.<anonymous closure> (package:polar_variety_sense_example/pages/polar_list_page.dart:172:21)
E/flutter (18923): <asynchronous suspension>
E/flutter (18923): 
```

Does `ERROR_INVALID_STATE` mean that the doFirstTimeUse() might not succeeded or do you have any suggestions what I'm doing wrong.

Some functions work though like setLocalTime or getDiskSpace.

This is my Kotlin implementation of doFirstTimeUse:

```kt
    private fun doFirstTimeUse(call: MethodCall, result: Result) {
        val arguments = call.arguments as Map<*, *>
        val identifier = arguments["identifier"] as? String
        val configMap = arguments["config"] as? Map<*, *>

        if (identifier == null || configMap == null) {
            result.error(
                "INVALID_ARGUMENTS",
                "Expected identifier and config map",
                null
            )
            return
        }

        val gender = configMap["gender"] as? String
        val birthDateString = configMap["birthDate"] as? String
        val height = (configMap["height"] as? Int)?.toFloat()
        val weight = (configMap["weight"] as? Int)?.toFloat()
        val maxHeartRate = configMap["maxHeartRate"] as? Int
        val vo2Max = configMap["vo2Max"] as? Int
        val restingHeartRate = configMap["restingHeartRate"] as? Int
        val trainingBackground = configMap["trainingBackground"] as? Int
        val deviceTime = configMap["deviceTime"] as? String

        // Validate required parameters
        if (gender == null || birthDateString == null || height == null || weight == null ||
            maxHeartRate == null || vo2Max == null || restingHeartRate == null ||
            trainingBackground == null || deviceTime == null
        ) {
            result.error(
                "INVALID_CONFIG",
                "Invalid configuration parameters",
                null
            )
            return
        }

        // Parse birth date
        val dateFormat = java.text.SimpleDateFormat("yyyy-MM-dd", java.util.Locale.getDefault())
        val birthDate = dateFormat.parse(birthDateString)
            ?: throw IllegalArgumentException("Invalid birth date format")

        // Map gender string to PolarFirstTimeUseConfig.Gender enum
        val genderEnum = when (gender) {
            "Male" -> PolarFirstTimeUseConfig.Gender.MALE
            "Female" -> PolarFirstTimeUseConfig.Gender.FEMALE
            else -> throw IllegalArgumentException("Invalid gender value")
        }

        // Create PolarFirstTimeUseConfig instance
        val ftuConfig = PolarFirstTimeUseConfig(
            genderEnum,
            birthDate,
            height,
            weight,
            maxHeartRate,
            vo2Max,
            restingHeartRate,
            trainingBackground,
            deviceTime
        )

        // Call the Polar API
        wrapper.api
            .doFirstTimeUse(identifier, ftuConfig)
            .subscribe({
                runOnUiThread { result.success(null) }
            }, {
                runOnUiThread {
                    result.error(it.toString(), it.message, null)
                }
            })
            .discard()
    }
```

rkangast, 2025-01-24:
Can you confirm that you're wearing the device while starting the measurement?

jimmyzumthurm, 2025-01-24:
@abbjetmus what are the optical heart rate sensors LED showing if not connected to the charger ? Are they showing the "Waiting for FTU" animation, see here ? https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/products/Polar360.md#waiting-for-first-time-use

abbjetmus, 2025-01-24:
@jimmyzumthurm Yes it looks like the animation for "Waiting for FTU", so my call might not succeed then?

rkangast, 2025-01-24:
Yes, I just tried FTU with a factory reseted 360 device. FTU was fine and I could start a measurement after FTU. Can you see the 360 battery status? Battery status has to be at least 10 percent, otherwise 360 will response with "ERROR_INVALID_STATE".

abbjetmus, 2025-01-24:
The battery is above 90% and I still can't get it working, tried on iOS as well but same problems...

rkangast, 2025-01-24:
This is how our app is calling FTU using RXJava,
 ```
api.doFirstTimeUse(deviceId, ftuConfig = polarFirstTimeUseConfig)
    .observeOn(AndroidSchedulers.mainThread())
    .subscribeOn(Schedulers.io())
    .subscribe({
        showMessage("Physical data sent successfully")
    }, { error ->
        showMessage("Error sending Physical data")
    })
 ```

Maybe this can help?

rkangast, 2025-01-24:
Oh yes it can, please upgrade to 5.7.12. There has been improvements implemented to doFirstTimeUse method so that it does not return too early.