polarofficial/polar-ble-sdk #543: Polar SDK 5.12.0 on iOS stuck in connecting... when doing connectToDevice

abbjetmus, 2025-01-26:
***Platform your question concerns:*** 
- [ ] Android 
- [ X] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Polar 360
- [ ] Device is not relevant for this question

**Description:**
Hi guys! Thank you for the help on this issue for doFirstTimeUse [https://github.com/polarofficial/polar-ble-sdk/issues/542](https://github.com/polarofficial/polar-ble-sdk/issues/542), upgrading to sdk version 5.12.0 fixed that issue. Unfortunetly I have another issue on iOS where I get stuck in connecting... state all the time when doing connectToDevice. Everything works fine on Android. I don't get any help from logs either. Do you have any suggestions how to fix that, or why that is?

```
  - PolarBleSdk (5.12.0):
    - Alamofire (~> 5.8.1)
    - RxSwift (~> 6.5.0)
    - SwiftProtobuf (~> 1.0)
    - Zip (~> 2.1.2)
```

[image]

Also another question I have. I feel like connecting to the sensors in general is problematic when doing from bakground jobs, get timeout when connecting and so on. I saw an old issue about this
[https://github.com/polarofficial/polar-ble-sdk/issues/82](https://github.com/polarofficial/polar-ble-sdk/issues/82)
should one use these function api.backgroundEntered(); and api.foregroundEntered(); for use case jumping back and forth between foreground and background?

abbjetmus, 2025-01-27:
@rkangast @jimmyzumthurm sry for tagging you guys, working on an approaching deadline and release and wonder if you have any feedback on this question?

jimmyzumthurm, 2025-01-27:
@abbjetmus Hi, I suspect something is going wrong with the pairing between your Polar 360 and iOS device. Please see the notice about BLE connectivity in this page -> https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/products/Polar360.md#ble-functionnality

To eliminate that suspicion, could you try to perform factory defaults to your device ? (That could be done by pressing needle button under the wrist band while being connected to charger or use the SDK command for it), remove the pairing on your iOS device and then reconnect to your device and do the pairing again ?

abbjetmus, 2025-01-27:
@jimmyzumthurm I have already done the factory reset from Android device since it works well there. But doesn't help, the lights blinks according to doFirstTimeUse after reset from Android. Can it be that this sensor Polar 360 has extended permission requirements on iOS?

abbjetmus, 2025-01-27:
Yes you are right, issue was with pairing, and one of my iphone behaved strange. My collegue said it helped removing the sensor from Flow website. Final question regarding the background job thing. 
I saw an old issue about this
[https://github.com/polarofficial/polar-ble-sdk/issues/82](https://github.com/polarofficial/polar-ble-sdk/issues/82)
should one use these function api.backgroundEntered(); and api.foregroundEntered(); for use case jumping back and forth between foreground and background?

rkangast, 2025-01-27:
Hi there,

Those two APIs have been deprecated and removed from the SDK and thus cannot be used anymore.