polarofficial/polar-ble-sdk #546: How do I switch device after being connected to another device

abbjetmus, 2025-01-29:
***Platform your question concerns:*** 
- [ ] Android 
- [X ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X ] Polar 360
- [ ] Device is not relevant for this question

**Description:**
Hi guys! It is very messy switching devices on iOS and generally difficult to connect with Polar 360. Most of the time something is preventing the connection flow.

Can you provide a successful flow that garanties a connection and FTU completion on iOS?

I connect the charger and do a factoryReset by pressing the small button on the sensor. After doing that I try to connect to the sensor from an Iphone and I still fail to connect, stuck in connecting. I try to remove the sensor from bluetooth settings on all devices and try again and it doesn't work. On Android it always works. Android tend to keep the pairings under bluetooth settings, but a remove it from there do factoryReset and still can't connect to iOS.

My collegue says that this always works, but can't be this tedious to get a connection...

1. Remove info about Polar sensor in Our app from previous tries.
2. Connect the Polar sensor with the charging cable and do a factory reset via the button on the top. Removes the cord when done.
3. Delete Our app.
4. Go into Iphone settings->Bluetooth and select "Forget" on the Polar sensor in the list.
5. Restarts the phone completely.
6. Install Our app via TestFlight.
7. Launch Our app
8. Connect to the sensor again.
9. Clicks OK when pairing.

jimmyzumthurm, 2025-01-29:
@abbjetmus I'm not too familiar with how iOS handles the pairing (I'll let my colleagues answer that), but what is important for sure is that both iOS phone and device have cleared their pairing info. 
If you remove the pairing on device side only (factory defaults) without doing it on the mobile side, I know that Android can handle that perfectly fine, make a new pairing request and all is good. iOS needs that the pairing is completely removed (not sure how that is done though) before a new one can be made. 

I'm wondering if step 4. somehow doesn't apply immediately if the connection is still alive between the 2 devices ? After step 2 is done could the app still reconnect to device thus the step 3 and 5 are needed ?

abbjetmus, 2025-01-29:
Step 4. doesn't seem to help. Step 5 seems to be necessary for it to work on iPhone. If 3 is done no reconnects will be attempted.

abbjetmus, 2025-01-30:
Closing this!

closed 2025-01-30 by abbjetmus

jimmyzumthurm, 2025-01-30:
@abbjetmus Did you find any easier way to do this ? (for my general culture)

abbjetmus, 2025-01-30:
No I didn't unfortunetly, seems that restart of the phone is easiest!