polarofficial/polar-ble-sdk #548: Offline recording errors on Polar 360. The size of the file is 39, accessing the index 40!

abbjetmus, 2025-01-30:
**Platform on which you observed the bug:** 
- [ X] Android 
- [ X] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Polar 360
- [ ] Device is not relevant for this bug

**Describe the bug**
Hi! When doing offline recording I get strange errors from the sensor, we only store PPI values offline. Not sure eactly where the main issue lies, but requests fail and when doing `getOfflineRecord` I get errors according to the logs section. Then probably the sensor end up in a error state and `getOfflineRecordingStatus` gives other errors as well. The Polar 360 has been inactive for a long time say 20 hours not sure and offline recording is active on the sensor but the sensor has not been on the arm.

Please why is that, have release next week?

**How to Reproduce**
I'm not fully sure how to reproduce this but our flow is:
Each time the app comes to foreground we connect to the sensor and do:
1. connectToDevice, after connected
2. getOfflineRecordingStatus
3. stopOfflineRecording
4. listOfflineRecordings
5. getOfflineRecord - we only store PPI offline
6. removeOfflineRecord - remove files when reading done
7. startOfflineRecording

**Expected behavior**
This should work  I think.

**Screenshots and logs**

**getOfflineRecord errors**

```error
com.polar.androidcommunications.api.ble.model.offlinerecording.OfflineRecordingError$OfflineRecordingErrorMetaDataParseFailed: Data payload size parse failed. The size of the file is 39, accessing the index 40```

AI explanation: 

This suggests that the Polar SDK is encountering an issue when trying to parse the data received from the device. Specifically, it is trying to access an index (40) that is out of bounds for the data payload (size 39).

**getOfflineRecordingStatus errors**

```error
W/artrest.app.de(15185): Long monitor contention with owner RxCachedThreadScheduler-14 (20616) at void com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.sendControlPointCommand$lambda$16(com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient, byte[], com.polar.androidcommunications.api.ble.model.gatt.client.pmd.PmdControlPointCommandClientToService, io.reactivex.rxjava3.core.SingleEmitter)(BlePMDClient.kt:257) waiters=0 in void com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.sendControlPointCommand$lambda$16(com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient, byte[], com.polar.androidcommunications.api.ble.model.gatt.client.pmd.PmdControlPointCommandClientToService, io.reactivex.rxjava3.core.SingleEmitter) for 13.354s
I/flutter (15185): API_OFFLINE - Failure processing offline data! 1 PlatformException(java.lang.Exception: Pmd response failed to receive in timeline, Pmd response failed to receive in timeline, null, null) #0      StandardMethodCodec.decodeEnvelope 
```
AI explanation: 

The error you're encountering suggests that there is a long monitor contention issue, which typically indicates that a thread is waiting for a long time to acquire a lock that is held by another thread. In your case, the thread RxCachedThreadScheduler-14 is holding a lock for an extended period, causing other threads to wait.

Additionally, the error message API_OFFLINE - Failure processing offline data! and the PlatformException indicate that the Polar SDK is failing to receive a response within the expected timeline when trying to process offline data.

Steps to Troubleshoot and Resolve the Issue
Check Threading and Synchronization:

Ensure that the getOfflineRecordingStatus method is not blocking the main thread. The subscribe method should be called on a background thread, and the result should be returned to the main thread using runOnUiThread.

Verify that the Polar SDK is not performing any long-running operations on the main thread.

Increase Timeout:

If the issue is related to a timeout, consider increasing the timeout duration for the operation. This can be done by configuring the Polar SDK or the RxJava scheduler.

jimmyzumthurm, 2025-01-30:
Hi @abbjetmus , 

This occurs when the SDK attempts to read a file that was created on the device side but never got any actual data inside it, only the file meta data which is written when the start command is received. This can happen when doing PPI offline recording but not wearing the sensor, as PPI might only get produced when wearing it. When receiving that error the file should get deleted. 

The SDK could handle this better and recognize that case as "no data available" instead of that error message. 

Ping @palmqvisti

abbjetmus, 2025-01-31:
I added a check to see if the record size is greather than 40 before trying to read the file. Since empty files seems to have a size of 39. But I still get some errors when file sizes are 117.

```error
API_OFFLINE - Failure processing offline data! 1 PlatformException(java.lang.IndexOutOfBoundsException: toIndex (11008) is greater than size (76)., toIndex (11008) is greater than size (76)., null, null) #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
```

rkangast, 2025-02-17:
Hi @abbjetmus,

I spent some time with this issue and I cannot reproduce the issue. I know that this kind of issue can happen if the incoming data from the device is for some reason corrupted. For example some air packets have dropped during transfer. That can cause this kind of error where the expected data chunk length does not match with the length of the incoming data. 

I will leave the code as it is, hiding the problem with some sanity check might lead to hiding possible future device or other sdk problems.

mikeread-dev, 2025-03-06:
I have seen this a couple of times, but only if I `startOfflineRecording` them immediately `listOfflineRecordings`.  It disappears once it has some data recorded.

palmqvisti, 2025-03-07:
Hi @abbjetmus, the device uses an internal buffer to record data samples before it writes them to a file. So, the file is not created immediately after the start of the recording, but after a delay. The delay can be up to five minutes depending on data type.