polarofficial/polar-ble-sdk #550: Problems with Polar 360 onboarding!

abbjetmus, 2025-02-05:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Polar 360
- [ ] Device is not relevant for this question

**Description:**

Hi Guys!
We are trying to provide a seamless onboarding for our customers using Polar 360, but unfortunetly the new flow on Polar 360 with FTU are troublesome.

Polar 360 allows us to be connected to the sensor - status = connected without having completed PAIRING or FTU, but we cannot start measuring before doing those two things. 
We also don't know if they succeeded because we can't ask the sensor about if FTU is done or if PAIRING is done. Our customers have problem with this since we want to start measuring directly. But they end up with one of these steps failing and our app don't know about it.

If they have succeeded with both and do a factory reset later I have a variable in shared preferences indicating if FTU is done which will not be valid anymore.

Can we ask the sensor with PAIRING is done or FTU is done?

It would have been nice if one could decide to skip FTU completely.

With Polar Verity Sense these issues was not present since it had an easier onboarding.

vedantgurav, 2025-02-05:
Hey. @abbjetmus, we have had similar issues with the pairing flow. On iOS we are handling 

We now look out for the [`PolarErrors.notificationNotEnabled`](https://github.com/polarofficial/polar-ble-sdk/blob/daa09ce411ab7ffc5bfc2832aa881f3c9e926fa8/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/api/errors/PolarErrors.swift#L8) error which means pairing has failed.

Missing FTU usually causes the [`PmdResponseCode.errorInvalidState`](https://github.com/polarofficial/polar-ble-sdk/blob/daa09ce411ab7ffc5bfc2832aa881f3c9e926fa8/sources/iOS/ios-communications/Sources/iOSCommunications/ble/api/model/gatt/client/pmd/PmdControlPointResponse.swift#L40) error, although this is not exclusive.

Additionally to handle cases where the Device is factory reset, we check [here](https://github.com/polarofficial/polar-ble-sdk/blob/daa09ce411ab7ffc5bfc2832aa881f3c9e926fa8/sources/iOS/ios-communications/Sources/iOSCommunications/ble/endpoints/corebluetooth/central/CBDeviceListenerImpl.swift#L205) and [here](https://github.com/polarofficial/polar-ble-sdk/blob/daa09ce411ab7ffc5bfc2832aa881f3c9e926fa8/sources/iOS/ios-communications/Sources/iOSCommunications/ble/endpoints/corebluetooth/central/CBDeviceSessionImpl.swift#L363) for the CoreBluetooth errors `.peerRemovedPairingInformation` and `.encryptionTimedOut` respectively. In this case we just prompt the user to manually forget the P360 from bluetooth settings. If we get `encryptonTimedOut`, a P360 reset is also required. After this the app can pair once more.

abbjetmus, 2025-02-05:
Thanks for the response @vedantgurav, looking forward for a response from Polar as well!

rkangast, 2025-02-10:
FLOW-62499

palmqvisti, 2025-02-10:
Hello from Polar! We will add a "isFtuDone" kind of method to check for the state of FTU. Hope that helps in this issue.

abbjetmus, 2025-02-10:
That would be nice also if there was a way to know if the sensor already has a pairing and to which device would be helpful.

rkangast, 2025-02-12:
Hi,

Unfortunately that cannot be done. Once Polar 360 has been paired to a device it will no longer pair to any another device that would possible allow checking the existing pairing data.

abbjetmus, 2025-02-12:
@rkangast we are on boarding new users next week is it possible to get a patch this week with isFtuDone?

rkangast, 2025-02-14:
Hello @abbjetmus,

isFtuDone method will be released today in the next SDK release (5.14.0).

abbjetmus, 2025-02-14:
Thank you @rkangast, you have the best support in the world!

rkangast, 2025-02-17:
Hello @abbjetmus,

Deployment of version 5.14.0 has been now  fixed. Sorry for the inconvenience!

abbjetmus, 2025-02-17:
@rkangast I don't think it's fixed still get latest as 5.13.0?

rkangast, 2025-02-18:
@abbjetmus Latest is 5.14.0 and should be found with CocoaPods, see https://github.com/CocoaPods/Specs/tree/master/Specs/1/d/4/PolarBleSdk/5.14.0

closed 2025-02-24 by abbjetmus