polarofficial/polar-ble-sdk #552: Connection timeout in background job Polar 360.

abbjetmus, 2025-02-12:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X] Polar 360
- [ ] Device is not relevant for this question

**Description:**

Hi! When using Polar 360 in our background job we timeout if we don't succeed with connectToDevice after 30 seconds. Since iOS don't allow longer jobs. The sensor don't provide any reason why it can't connect. Suspect but not sure it might already have a connection to the device. But we do at the end a disconnect in or try - catch - finally -> disconnect.

Looking at this old issue, it's mentioned to first scan for devices to check if it's advertising and if so check for device info isConnectable.
[https://github.com/polarofficial/polar-ble-sdk/issues/286](https://github.com/polarofficial/polar-ble-sdk/issues/286)

Do you have any suggestions how to approach this situation?

Was thinking I might start with a disconnect first to make sure it's not connected.

abbjetmus, 2025-02-12:
So the issue seems to be in general that the polar sdk is not working in a background job since doing `polar.searchForDevice()` also gives timeout. I'm I missing something when working with polar in a background job?

rkangast, 2025-02-14:
Hi,

Could it be that you have connected the sensor to another mobile device? The Polar 360 allows only one Bluetooth connection. When paired once, it will remember the pairing and will refuse any new pairing to any other device. 

You can remove pairing from Polar 360 by using [doFactoryReset-api](https://github.com/polarofficial/polar-ble-sdk/blob/5.13.0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.kt#L308) with preservePairingInformation flag set to false. Note, that factory reset will erase any measurement data and return the sensor device to factory state.

abbjetmus, 2025-03-19:
@rkangast I'm re-opening this. I have problems completing our offline recording import flow in a background job. This always succeeds in the foreground:

This is Flutter code, but I don't think the issue is on the flutter side, showing the code so you can see the flow:
```dart

final sensorHandler = BackgroundJobSensorHandlerPolar(identifier, Polar());

    final sensor = await sensorHandler.connect(const Duration(seconds: 30));
    debugPrint('Connect Completed');

    try {
      await sensor.sdkFeatureReady.firstWhere(
        (e) =>
            e.identifier == identifier &&
            e.feature == PolarSdkFeature.offlineRecording,
      );

      debugPrint('SDK Feature Ready Completed');

      final offlineRecordingsType =
          await sensor.getOfflineRecordingStatus(identifier);

      if (offlineRecordingsType.isEmpty) {
        debugPrint('Ingen offline status att stoppa');
      }

      for (final recordingType in offlineRecordingsType) {
        if (recordingType == PolarDataType.ppi) {
          await sensor.stopOfflineRecording(
            identifier,
            PolarDataType.ppi,
          );
          debugPrint('Stoppade PPI data');
        }
      }

      var recordings = await sensor.listOfflineRecordings(identifier);

      debugPrint('recordings: ${recordings.length}');

      for (final record in recordings) {
        debugPrint(
          'Processing offline data: ${record.path}, size: ${record.size}',
        );
        if (record.type == PolarDataType.ppi && record.size > 40) {
          try {
            final ppiData = await sensor.getOfflinePpiRecord(
              identifier,
              record,
            );

            if (ppiData != null) {
              debugPrint('ppiOfflineListLength ${record.path}');

              // Use a for loop instead of forEach
              for (int i = 0; i < ppiData.data.samples.length; i++) {
                final s = ppiData.data.samples[i];
                // Only log every 60th value (approximately one per minute)
                debugPrint(
                  'PPI: ${s.ppi}, HR: ${s.hr}, Time: ${s.timeStamp.toIso8601String()}',
                );
              }
            }
            // Delete the record after successful processing
            await sensor.removeOfflineRecord(identifier, record);
          } catch (e) {
            debugPrint('Error processing record: $e');
          }
        }
      }
[... 119 lines cut ...]
E/AndroidRuntime(26586): 	at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4855)
E/AndroidRuntime(26586): 	at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run(SingleSubscribeOn.java:89)
E/AndroidRuntime(26586): 	at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
E/AndroidRuntime(26586): 	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
E/AndroidRuntime(26586): 	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
E/AndroidRuntime(26586): 	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
E/AndroidRuntime(26586): 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:307)
E/AndroidRuntime(26586): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
E/AndroidRuntime(26586): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
E/AndroidRuntime(26586): 	at java.lang.Thread.run(Thread.java:1012)
E/AndroidRuntime(26586): Caused by: com.polar.sdk.api.errors.PolarDeviceDisconnected
E/AndroidRuntime(26586): 	at com.polar.sdk.impl.BDBleApiImpl.handleError(BDBleApiImpl.kt:4318)
E/AndroidRuntime(26586): 	at com.polar.sdk.impl.BDBleApiImpl.access$handleError(BDBleApiImpl.kt:139)
E/AndroidRuntime(26586): 	at com.polar.sdk.impl.BDBleApiImpl$listFiles$2.apply(BDBleApiImpl.kt:3787)
E/AndroidRuntime(26586): 	at com.polar.sdk.impl.BDBleApiImpl$listFiles$2.apply(BDBleApiImpl.kt:3786)
E/AndroidRuntime(26586): 	at io.reactivex.rxjava3.internal.operators.flowable.FlowableOnErrorNext$OnErrorNextSubscriber.onError(FlowableOnErrorNext.java:94)
E/AndroidRuntime(26586): 	... 40 more
I/flutter (26586): Device disconnected
I/Process (26586): Sending signal. PID: 26586 SIG: 9
Lost connection to device.

Exited.
```

Please do you have any suggestions here?

abbjetmus, 2025-04-07:
Hi! I'm reopening this issue since it still happens often especially in a background job. Even if the background job tries to connect it is the same device as the foreground process is connected to. Although we disconnect before ending up in the background.

rkangast, 2025-04-10:
Hi, it is probable that listing the offline files takes longer than the iOS background job timeout (30sec). And thus the sensorHandler.disconnect() will get called while the listing process is still ongoing. Is it possible to move this sequence to a foreground process?

rkangast, 2025-07-04:
This issue is getting stale, thus I will close this issue. Please, do reopen when needed.

closed 2025-07-04 by rkangast