polarofficial/polar-ble-sdk #554: Apple require Polar BLE SDK with newer version of Alamofire (5.9 or more)

undertaker1753, 2025-02-16:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [X] Device is not relevant for this bug

**Describe the bug**
Apple require Polar BLE SDK with newer version of Alamofire (5.9 or more)

**How to Reproduce**
a)
    pod 'PolarBleSdk', '>= 5.11'
    pod 'Alamofire', '>= 5.9'

pod update

-> no update possible

b)
    pod 'PolarBleSdk', '>= 5.11'

pod update

-> no upload to App Store possible

**Expected behavior**
update Alamofire, export to App Store

**Screenshots and logs**

a)
pod update PolarBleSdk
Updating local specs repositories
Analyzing dependencies
[!] CocoaPods could not find compatible versions for pod "Alamofire":
  In snapshot (Podfile.lock):
    Alamofire (= 5.10.2, >= 5.9)

  In Podfile:
    Alamofire (>= 5.9)

    PolarBleSdk (>= 5.11) was resolved to 5.13.0, which depends on
      Alamofire (~> 5.8.1)

b)

TMS-91061: Missing privacy manifest - Your app includes “Frameworks/Alamofire.framework/Alamofire”, which includes Alamofire, an SDK that was identified in the documentation as a commonly used third-party SDK. Starting February 12, 2025, if a new app includes a commonly used third-party SDK, or an app update adds a new commonly used third-party SDK, the SDK must include a privacy manifest file or you won’t be able to submit the app for review in App Store Connect. Please contact the provider of the SDK that includes this file to get an updated SDK version with a privacy manifest. For more details about this policy, including a list of SDKs that are required to include signatures and manifests, visit: https://developer.apple.com/support/third-party-SDK-requirements.

https://github.com/Alamofire/Alamofire/pull/3792

pth-pe-gh, 2025-02-17:
FLOW-62638

undertaker1753, 2025-02-18:
what is the meaning of your answer?

pth-pe-gh, 2025-02-19:
@undertaker1753  We are working on the issue. The code is simply a reference for Polar internal tracking of this issue. Thanks for bringing this to our attention; we target to fix this in next release.

pth-pe-gh, 2025-02-26:
@undertaker1753 latest SDK iOS release has removed the Alamofire dependency; this should no longer block apps built based on SDK. Please file new bug if similar issue is raised for SDK itself.

pth-pe-gh, 2025-02-28:
Podspec update was left out from release for some reason. Reopening.

pth-pe-gh, 2025-03-05:
SDK version 5.16.0 has been released, with more complete cleanup of Alamofire. Note that example app update is scheduled separately; it still contains Podfile.lock reference but SDK should be clean. 

@undertaker1753 - can you please verify SDK side removal is complete with version 5.16.0 and working with your app. Also, let us know if your app still gets rejected with missing privacy info due SDK inclusion.

closed 2025-03-06 by undertaker1753