polarofficial/polar-ble-sdk #555: BLE application will not start the recording on my h10 - this h10 will send bluetooth to Polar Sensor Logger and Polar Beat

smyffanon, 2025-02-16:
**Platform on which you observed the bug:** 
- [ x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Cannot start in-memory recording

**How to Reproduce**
do not know how this state happened to the h10

**Expected behavior**
clicking on start recording should start the recording

**Screenshots and logs**
This issue  below has re-appeared: 
https://github.com/polarofficial/polar-ble-sdk/issues/453

The phone connects OK,
the h10 reports zero recordings are currently on the h10.
the h10 reports there is no on-going recording currently happening.

The last time it happened, a new firmware update appeared for the h10's firmware, 
and when I re-flashed the device to the new firmware the issue disappeared.

Currently there's no way to re-flash and there's no new version to try the same fix as last time.

I'm hoping there's another way to fix that was not available the last time. 

Maybe there's some other app that will allow re-flashing the h10?

pth-pe-gh, 2025-02-18:
FLOW-62642

pth-pe-gh, 2025-02-18:
@smyffanon - we can not reproduce this using SDK example app and H10 with Firmware version 3.3.1

pth-pe-gh, 2025-02-18:
@smyffanon  - we can not reproduce this issue using H10 with latest firmware 3.3.1 and iOS example app.

pth-pe-gh, 2025-02-19:
@smyffanon - if you still experience this issue, please contact Polar care for H10 related support. As we have not been able to reproduce this yet, it seems this is not a bug in SDK at the moment.

closed 2025-02-19 by pth-pe-gh