polarofficial/polar-ble-sdk #556: Accessing offline recording using python

jukaradayi, 2025-02-18:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hello, 

I was wondering if it was possible to retrieve offline recordings from a polar H10. 
Using the bleak library I was able to stream ECG Data, and start recording. I was also able to compile a protobuf GET request and send it to the device, but I haven't managed to get any response. Is it possible to access offline recordings using python, or does it only work using the SDK ?
Thanks in advance,

palmqvisti, 2025-02-19:
Currently we do not support accessing offline records using Python.

jukaradayi, 2025-02-20:
Thank you for your quick response !
One last question : Is it possible to do offline recording of ECG measurements on the polar H10, or is it only possible to record HeartRate offline ? In the Android SDK I only found the Heart rate option.. 
Thanks in advance

palmqvisti, 2025-02-25:
You can record one training session with H10 into internal memory and download it with SDK. Functionality is similar to what is in Polar Beat app.

palmqvisti, 2025-02-27:
Offline recording of ECG is not supported for H10.

closed 2025-07-04 by rkangast