polarofficial/polar-ble-sdk #558: setLocalTime becomes problematic sometimes

abbjetmus, 2025-02-24:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ X] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X] Polar 360
- [ ] Device is not relevant for this question

**Description:**
Hi!

We are experiencing some time issues when reading offline data.
We only use PPI offline.

My worfklow is as follows.

On onboarding I call setLocalTime: `await polar.setLocalTime(sensorId, DateTime.now());``
 And StartOfflineRecording for PPI.

 Each time we connect again we do: 
1. StopOfflineRecording

2. Read each recording data.

3. Delete each record.

4. Upload the data

5. SetLocalTime again

6. StartOfflineRecording

Sometimes the data get shifted for hours like from this barchart. Here the data should not have a gap instead the first part should end where the second part start.

[image]

I'm I doing something wrong with this approach shouldn't the offline data always start from setLocalTime?

abbjetmus, 2025-02-25:
@rkangast do you happen to have some feedback on this?

rkangast, 2025-02-25:
@abbjetmus I created a task for us to add timestamp to PPi data.

FLOW-6280

abbjetmus, 2025-02-25:
Thank you @rkangast do know when we can expect it to be completed?

abbjetmus, 2025-02-25:
Great, just to be clear when I call setLocalTime I expect the PPI samples to include timestamp according to that time.

jimmyzumthurm, 2025-02-25:
@abbjetmus That is correct. The timestamp of the data will be in system time though (UTC time), so you will need to add/subtract the timezone offset to convert it to local time. But I believe that the setLocalTime() API sets both local time and system time inside the device. @rkangast can correct me. 

Read about timestamps here : https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/TimeSystemExplained.md

Grefw3n, 2025-02-26:
> [@abbjetmus](https://github.com/abbjetmus) I created a task for us to add timestamp to PPi data.
> 
> FLOW-6280

What's the time estimate for this?

closed 2025-07-04 by rkangast