polarofficial/polar-ble-sdk #560: Polar 360 - getOfflineRecordingStatus error when temp is recording

Marcus122, 2025-02-25:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X] Polar 360
- [ ] Device is not relevant for this bug

**Describe the bug**
When we have started recording temperature and call getOfflineRecordingStatus we then get the error 
Error when map measurement type TEMPERATURE to polar feature

Looking through the Kotlin code 

[image]

line 140 being mapped from TEMPERATURE to TEMPERATURE

but 156 being mapped from OFFLINE_TEMP to TEMPERATURE (OFFLINE_TEMP should just be TEMPERATURE I think?)

**How to Reproduce**
Start offline recording for temperature and then call getOfflineRecordingStatus

pth-pe-gh, 2025-02-26:
@Marcus122 Thanks for reporting, you experienced this with latest SDK version v5.15.0 released on Monday?

FLOW-62820

rkangast, 2025-02-26:
Hello @Marcus122 

This issue has already been fixed in release 5.15, see [PolarDataUtils](https://github.com/polarofficial/polar-ble-sdk/blob/5.15.0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/utils/PolarDataUtils.kt#L307)

Marcus122, 2025-02-26:
Brill thanks @rkangast . Have tested on the new release and all works great.