polarofficial/polar-ble-sdk #563: Problem update Polar360 firmware to 1.2.11

janbarari, 2025-02-28:
**Platform on which you observed the bug:** 
- [x] Android 

**Device on which you observed the bug:**
- [x] Polar 360

**Describe the bug**
When I try to update firmware to version 1.2.11 device keep disconnected after `Preparing device for firmware update`.

**How to Reproduce**
Current SDK: 5.15.0
Current Firmware: 1.1.80
Is firm time use set: true
Is device time set: true
Is device settings set: true
is tried before and after factory reset: true

**Expected behavior**
It should update the firmware but it doesn't

pth-pe-gh, 2025-02-28:
Hi @janbarari - Firmware update from 1.1.80 to 1.2.11 works in our tests. We have not been able to reproduce the reported behaviour. Is there any additional information that you could share, such as 
- Has firmware update worked for you with previous SDK version using the same Polar 360 and same Android phone?  
- With same phone and Polar 360, does your app code complete all previous operations with the sensor properly before proceeding with firmware update? 

FLOW-62863

pth-pe-gh, 2025-02-28:
@janbarari - in 1.1.80 firmware, there is relatively low limit on how many resets one can perform within certain time period before Polar 360 restores to factory defaults. Factory defaults also removes pairing. Possibly your bug was caused by too many resets combined before and during firmware update. Please try firmware update again so that there has been at least 5 minutes from previous reset operation. Let us know if it helps.

janbarari, 2025-02-28:
@pth-pe-gh
Thank you for your feedback! The issue wasn’t related to a factory reset. I found that when I turned the Polar SDK's automatic connection OFF, the firmware update couldn't complete. Once I turned it ON, the update went through successfully.

I wanted to share this in case you'd like to adjust the SDK behavior or if anyone else encounters the same issue.

Best,
MJ

closed 2025-02-28 by janbarari