polarofficial/polar-ble-sdk #565: Update RxSwift Dependency to ~> 6.8.0 to Address Privacy Manifest Requirement from Apple

takuta98, 2025-03-03:
Hello,

I am developing an iOS app that integrates with Polar sensors using PolarBleSdk. Currently, the podspec specifies:

`s.dependency 'RxSwift', '~> 6.5.0'`

However, our project requires RxSwift version ~> 6.8.0 to meet Apple's latest review guidelines. We have encountered a dependency conflict during pod install when our Podfile specifies RxSwift ~> 6.8.0.

### Issue Details
Current PolarBleSdk Podspec: RxSwift dependency is set to ~> 6.5.0
Our Requirement: RxSwift ~> 6.8.0, as mandated by Apple's guidelines
Problem: The version conflict prevents us from using the required RxSwift version, forcing us to either downgrade or fork the repository. This is not ideal, as we need to integrate PolarBleSdk for Polar sensor connectivity.

### Apple's Feedback
We received the following message during the app review process:

> ITMS-91061: Missing privacy manifest - Your app includes “Frameworks/RxSwift.framework/RxSwift”, which includes OneSignalOutcomes, an SDK that was identified in the documentation as a commonly used third-party SDK. If a new app includes a commonly used third-party SDK, or an app update adds a new commonly used third-party SDK, the SDK must include a privacy manifest file. Please contact the provider of the SDK that includes this file to get an updated SDK version with a privacy manifest. For more details, visit: [Apple Third-Party SDK Requirements](https://developer.apple.com/support/third-party-SDK-requirements).

### Request
Would it be possible to update the RxSwift dependency in the PolarBleSdk podspec to support RxSwift ~> 6.8.0 (or a compatible range that includes 6.8.0)? We have tested our integration with RxSwift 6.8.0 and found no compatibility issues with PolarBleSdk.

This update would allow us to comply with Apple’s review guidelines while continuing to use PolarBleSdk for Polar sensor integration.

Thank you for your time and consideration.

TeemuLauPolar, 2025-03-06:
FLOW-62985

medlay, 2025-03-21:
In the latest update (5.17.0) `s.dependency 'RxSwift', '~> 6.5.0'`. Therefore, it's impossible to update the app on the App Store.

closed 2025-06-26 by pth-pe-gh

TaskLocal, 2025-06-26:
@pth-pe-gh I updated the Package.swift file for this ^ in https://github.com/polarofficial/polar-ble-sdk/pull/636