polarofficial/polar-ble-sdk #566: Unstable Offline Recording on New Firmware (1.2.11)

janbarari, 2025-03-03:
**Platform on which you observed the bug:** 
- [x] Android 

**Device on which you observed the bug:**
- [x] Polar360

**Describe the bug**
ACC offline datatype recording couldn't start after some time in new Firmware update.

SDK: 5.15.0
Firmware: 1.2.11
Battery: ~70%

**How to Reproduce**
Connect to device, start ACC, HR, PPI offline recording together. disconnect and connect to sensor next hour and get the data. this loop should continue working always but after 5-6 hours, the starting ACC getting failure

TeemuLauPolar, 2025-03-06:
FLOW-62986

janbarari, 2025-03-12:
> FLOW-62986

Can you please tell me what is the meaning of this?

janbarari, 2025-03-18:
Hi
Is there any update regarding to my report?

samulimaa, 2025-03-18:
Hi, reply with FLOW-* prefix means that we have internally created ticked from that issue and it's on or backlog for further investigation.

samulimaa, 2025-03-27:
What extract error did you receive when trying to start ACC recording?

palmqvisti, 2025-05-19:
@janbarari are you still seeing this issue with the latest SDK 6.3.0? If it occurs, what is the exception exactly?

janbarari, 2025-06-10:
Issue is resolved.

closed 2025-06-10 by janbarari