polarofficial/polar-ble-sdk #567: The PolarEcgDataSample.getVoltage() method is not available in 5.15.0 version

amart384, 2025-03-05:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Related to the ECG stream. Up to version 5.13.0 in the subscriber, you could collect the data in a list and then process the voltage and timestamp fields, for each list element. In version 5.15.0 it seems that only the timestamp field can be access.

**How to Reproduce**
.subscribe(
	PolarEcgDataSample -> {
		for (PolarEcgDataSample data : PolarEcgDataSample.getSamples()){
				data.getVoltage(); //error not defined
				data.getTimeStamp();
		}
	}

**Expected behavior**
I don't know if in the new versions it is only allowed to subscribe and collect one data, one by one, with ECGSample Class. It would be useful if, in the same subscriber event, you could collect several data in a list, as it worked in version 5.13.0. It would be great if the number of values ​​to collect would be 64 . For example (64*130Hz= 492ms for each subscriber event).

Thank you.

TeemuLauPolar, 2025-03-06:
FLOW-62987

pth-pe-gh, 2025-03-07:
@amart384 - thanks for bug report. Here's some example Kotlin code how you can get the voltage using SDK now:

```
private fun printEcgLiveData(polarEcgData: PolarEcgData) {
         when (val ecgSample = polarEcgData.samples.last()) {
          is EcgSample -> {
              Log.d("DEBUG","Ecg uV:" + ecgSample.voltage.toString())
          }
          else -> {
              Log.d("DEBUG","Unhandled Ecg sample type")
          }
      }
}
```

So you need to cast the sample to appropriate PolarEcgDataSample subclass, EcgSample, to get the voltage value.

https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/model/EcgSample.html

amart384, 2025-03-08:
Thank you very much. It works.

closed 2025-03-08 by amart384