polarofficial/polar-ble-sdk #570: Array IndexOut OfBounds Exception

amart384, 2025-03-12:
**Platform on which you observed the bug:** 
- [X ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Version 5.13.0
An ArrayIndexOutOfBoundsException exception is generated when accessing a nonexistent position of an element

Exception io.reactivex.rxjava3.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | java.lang.ArrayIndexOutOfBoundsException: length=1; index=1
  at io.reactivex.rxjava3.plugins.RxJavaPlugins.onError (RxJavaPlugins.java:372)
  at io.reactivex.rxjava3.internal.util.AtomicThrowable.tryAddThrowableOrReport (AtomicThrowable.java:52)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError (FlowableFlatMap.java:564)
  at io.reactivex.rxjava3.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError (FlowableFlatMap.java:636)
  at io.reactivex.rxjava3.internal.operators.single.SingleToFlowable$SingleToFlowableObserver.onError (SingleToFlowable.java:68)
  at io.reactivex.rxjava3.internal.observers.ResumeSingleObserver.onError (ResumeSingleObserver.java:51)
  at io.reactivex.rxjava3.internal.operators.single.SingleDoOnSuccess$DoOnSuccess.onError (SingleDoOnSuccess.java:65)
  at io.reactivex.rxjava3.internal.operators.single.SingleObserveOn$ObserveOnSingleObserver.run (SingleObserveOn.java:79)
  at io.reactivex.rxjava3.android.schedulers.HandlerScheduler$ScheduledRunnable.run (HandlerScheduler.java:123)
  at android.os.Handler.handleCallback (Handler.java:958)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loopOnce (Looper.java:230)
  at android.os.Looper.loop (Looper.java:319)
  at android.app.ActivityThread.main (ActivityThread.java:9063)
  at java.lang.reflect.Method.invoke
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:588)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1103)
  
Caused by java.lang.ArrayIndexOutOfBoundsException: length=1; index=1
  at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.PmdMeasurementType$Companion.fromByteArray (PmdMeasurementType.kt:41)
  at com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.readFeature$lambda$19 (BlePMDClient.kt:325)
  at io.reactivex.rxjava3.internal.operators.single.SingleCreate.subscribeActual (SingleCreate.java:40)
  at io.reactivex.rxjava3.core.Single.subscribe (Single.java:4855)
  at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run (SingleSubscribeOn.java:89)
  at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run (Scheduler.java:644)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run (ScheduledRunnable.java:65)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call (ScheduledRunnable.java:56)
  at java.util.concurrent.FutureTask.run (FutureTask.java:264)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run (ScheduledThreadPoolExecutor.java:307)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1145)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:644)
  at java.lang.Thread.run (Thread.java:1012)

**How to Reproduce**
It is not known how to reproduce the exception

**Expected behavior**
Control to ensure that the exception is not generated, especially that the exception does not cause the app to crash.
Our upper-layer code will have the recovery code to a stable state.

Thankyou

samulimaa, 2025-03-18:
Hi, we have created internal ticket from this for further investigation. FLOW-63204

amart384, 2025-03-18:
Hello,

We don't know the firmware for the H10 sensor. We only know that four users have been affected. Three users are running Android 14 and one user is running Android 15, on the following devices: Google Akita, Samsung A15x, Samsung A54x, Samsung R11q

Thank you.

closed 2025-05-15 by amart384