polarofficial/polar-ble-sdk #571: NullPointerException on ConnectionHandler,  connectionInterface.setPhy()

amart384, 2025-03-12:
**Platform on which you observed the bug:** 
- [ X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
An exception is thrown when accessing a null element.
This exception is likely due to a bug in our code or incorrect usage of library calls.
However, I would appreciate it if you could ensure that the method is called only when the object is not null, in order to avoid the exception and prevent the application from crashing.

   
  Exception java.lang.NullPointerException:
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$1.setPhy (BDDeviceListenerImpl.java:273)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.connection.ConnectionHandler.connecting (ConnectionHandler.kt:276)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.connection.ConnectionHandler.commandState (ConnectionHandler.kt:135)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.connection.ConnectionHandler.servicesDiscovered (ConnectionHandler.kt:115)
  at com.polar.androidcommunications.enpoints.ble.bluedroid.host.GattCallback.onServicesDiscovered$lambda$4 (GattCallback.kt:72)
  at io.reactivex.rxjava3.internal.observers.CallbackCompletableObserver.onComplete (CallbackCompletableObserver.java:44)
  at io.reactivex.rxjava3.internal.operators.completable.CompletableObserveOn$ObserveOnCompletableObserver.run (CompletableObserveOn.java:89)
  at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run (Scheduler.java:644)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run (ScheduledRunnable.java:65)
  at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call (ScheduledRunnable.java:56)
  at java.util.concurrent.FutureTask.run (FutureTask.java:264)
  at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run (ScheduledThreadPoolExecutor.java:307)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1145)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:644)
  at java.lang.Thread.run (Thread.java:1012)

**Expected behavior**
To the extent possible, add a check similar to:
if (connectionInterface != null) {
  connectionInterface.setPhy(session);
}

Thank you

samulimaa, 2025-03-18:
Hi, we have created internal ticket from this for further investigation. FLOW-63205.

Could you tell version of the SDK and H10 firmware version you are using? Same versions apply to #570?

amart384, 2025-03-18:
Hello, 

The version of SDK 5.13.0. We don't know the firmware version since the exception text is taken from the Google Play console.

We only know that Android 13 and Android 14 have occurred. There is no record of other versions.

Thank you for your answer.

palmqvisti, 2025-04-22:
@amart384, can you provide more details about what use case or method call causes this exception?

closed 2025-05-19 by palmqvisti