polarofficial/polar-ble-sdk #572: Pod Update version 5.16.0

hamletAppsfactory, 2025-03-12:
**Platform on which you observed the bug:** 
- [ ] Android 
- [X] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [X] Device is not relevant for this bug

**Describe the bug**
Version 5.16.0 is not getting build

**How to Reproduce**
Update pod version to 5.16.0, try to build the project.

**Expected behavior**
Should successfully build the project 

**Screenshots and logs**

[image]

pth-pe-gh, 2025-03-18:
@hamletAppsfactory - that looks like a build error dependent on using Swift6. Are you using Swift6 in your project?

hamletAppsfactory, 2025-03-19:
@pth-pe-gh After some refresh and relaunch it works.

closed 2025-03-19 by hamletAppsfactory