polarofficial/polar-ble-sdk #573: 2 seconds time difference on Polar H10 when using setLocalTime

nvd-ai, 2025-03-13:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
When using setLocalTime on Android phone to set Polar H10 time, the difference between Polar clock and phone clock is 1-2 seconds. Normally this should be within 50-200 milliseconds which accounts for the time it takes for the message to travel from Phone and gets applied on Polar including network latency.

**How to Reproduce**
1. From Android set Polar H10 time:
`
val calNow = Calendar.getInstance(TimeZone.getTimeZone("UTC"))
updateTimeDisposable.set(
    polarBleApi
        ?.setLocalTime(deviceId, calNow)
        ?.subscribeOn(Schedulers.io())
        ?.subscribe(
            { log.d(TAG, "updateTime: setLocalTime finished") },
            { log.e(TAG, it.message, it) }))
`

2. Compare the time between Polar and Android device. This is done by marking the time on both devices physically. One marking method is using Accelerometer on the phone and Polar, bump the two devices quickly, and plot the accelerometer signal. You'll see a peak in accel on both devices. That's the physical source of time.  Then compare the timestamps from Polar and phone at that point in time.

**Expected behavior**
Time  difference to be within 50-200ms.

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

nvd-ai, 2025-03-13:
In practice I use a different method for getting the physical source of time, which is sending flashlight to polar ECG and another device and compare their times.

samulimaa, 2025-03-18:
Hi, we have created internal ticket from this for further investigation. FLOW-63206.

pth-pe-gh, 2025-04-07:
Related, please see https://github.com/polarofficial/polar-ble-sdk/issues/580