polarofficial/polar-ble-sdk #574: Clarification on Removing Data from Device Beyond removeOfflineRecord

abPlus21, 2025-03-14:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Other
- [ ] Device is not relevant for this question

**Description:**
Hello Polar BLE SDK Team,

I’ve been using the removeOfflineRecord method to delete offline records from the device (Polar360). However, I’ve observed that while the files seem to be removed, not all the occupied memory is freed. This makes me suspect that there might be some caching mechanism in place or that directories are not being removed along with the files. If I reset the wristband to factory settings.

Could you please clarify if there’s an additional function or step required to fully free up the memory on the device? Or is there something like caching that might be causing this behavior?

Thank you for your help!

pth-pe-gh, 2025-03-25:
@abPlus21 - there indeed was an issue that left empty offline recording folders on device (FLOW-63096). This issue should be now fixed in Android SDK 5.17.0. Please let us know if you still experience memory not being freed after updating to SDK 5.17.0.

abPlus21, 2025-03-28:
@pth-pe-gh - tested it on a couple of devices the past days. It definetly improved but all four devices are filled up after 4 days to 10-13%. I em certain that the files are completly removed, at least as far as I can see via listOfflineRecordings (that returns an empty list). 
Maybe if you could check if there are files created for some sort of logging, or for the input used for startOfflineRecording (e.g maybe the settings or the secret (which we actually do not use) is stored/cached in a seperate file that gets new entries for new startOfflineRecording)?

pth-pe-gh, 2025-03-28:
@abPlus21  - could you share the exact actions you do with the device, along with memory stats increase over time; that would help us reproduce this more precisely?  Include all method SDK calls in order if possible?
And yes, there is data that gets stored on device with active use and data deletion methods available that should help freeing up the memory. You can try those first. You can use `deleteStoredDeviceData` method to delete data by type; or `deleteDeviceDateFolders` method to remove all data stored by date.

abPlus21, 2025-04-02:
@pth-pe-gh - we call doFirstTimeUse (only for FTU), startOfflineRecording, stopOfflineRecording, getOfflineRecordingStatus, listOfflineRecordings, getOfflineRecord, removeOfflineRecord. This we do for acc, hr and ppi ~ we use also polarsensorsetting to set the respective recording settings
We get the info about remaining storage via getDiskSpace
Furthermore we use searchForDevice, connectToDevice, setApiLogger, setApiCallback, foregroundEntered, searchForDevice
(disconnectFromDevice)

We loop through startOfflineRecording, stopOfflineRecording, getOfflineRecordingStatus, listOfflineRecordings, getOfflineRecord, removeOfflineRecord. like every 1 to 6h. 

deleteStoreDeviceData works only on activity data and such but not on raw data?/ tested the deleteDeviceDateFolders but did not help

orestesgaolin, 2025-04-02:
Just two cents from us as we maintain a fork of the SDK with the implementation of removeOfflineRecord that works for us.

We noticed that the official implementation can leave empty directories even though the recording has been deleted successfully. When listing files empty directories are omitted but the FTP service still iterates over them making the listing longer, using more battery and eventually after 30 days leading to about 2 min long listing.

We reverted to hybrid of older implementation (for Verity Sense) and for Polar 360 we make sure to remove parent directory if there are no other files left. You can see the implementation [here](https://github.com/makevisible/polar-ble-sdk/blob/a0cb981179f91f456919388d851ba9071cb0283e/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/BDBleApiImpl.kt#L1795).

[image]

pth-pe-gh, 2025-04-02:
Reopening FLOW-63096 to check for missing corner cases in recursively removing parent directories when deleting offline recordings. Thanks @orestesgaolin for sharing your implementation. The current cases we know and should check against are:

- folder contains single offline recording -> delete recording + folder
- folder contains multiple offline recordings, one file for each type of recording -> delete only single file for the recording that was chosen to be deleted
- folder contains single type offline recording split to multiple files -> delete all files and the folder
- folder contains multiple offline recordings, each split to multiple files -> delete all split files for the recording that was chosen to be deleted
- folder contains multiple offline recordings, some of which are single file, other that are split to multiple files -> delete only file or files belonging to recording that was chosen to be deleted

orestesgaolin, 2025-04-02:
That looks correct!

Perhaps make sure that if you remove offline recording on a previous day then entire path to the recording is deleted e.g. `/U/0/20250401/R/235500/` as well as the `/U/0/20250401/R/` folder if you invoke this on April 2.

pth-pe-gh, 2025-04-02:
That is good observation since any unnecessary empty directory while listing recordings can contribute to the delay.

mikeread-dev, 2025-04-17:
I seem to get this error a few times a day: 
```
I/flutter (32248): Step 6: Deleting offline recordings...
I/flutter (32248): Deleting entry from 2025-04-17 21:38:37.000Z
I/System.out(32248): [PolarPlugin] Starting safeRemoveOfflineRecord for identifier: E3F31026, path: /U/0/20250417/R/143837/PPI.REC
I/System.out(32248): [PolarPlugin] safeRemoveOfflineRecord completed successfully
I/flutter (32248): Deleting entry from 2025-04-17 21:38:39.000Z
I/System.out(32248): [PolarPlugin] Starting safeRemoveOfflineRecord for identifier: E3F31026, path: /U/0/20250417/R/143839/HR.REC
I/System.out(32248): [PolarPlugin] safeRemoveOfflineRecord completed successfully
I/flutter (32248): Cleanup of offline recordings completed
I/System.out(32248): [PolarPlugin] Caught error in safeRemoveOfflineRecord RxJava chain: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | java.lang.Exception: NO_SUCH_FILE_OR_DIRECTORY
```
would this be something that is causing my memory to be eaten up slowly over a few weeks?  I did a factory reset when my lack of memory was causing issues, now it is slowly creeping back down: 
```
I/flutter (32248): Disk space: 15024128
I/flutter (32248): Disk space: 6545408
```
I'm on FW version 1.2.11 and v5.17.0 of the SDK.

pth-pe-gh, 2025-04-22:
@mikeread-dev the error log entry is known to us and related to Polar internal in-progress issue tickets for SDK (FLOW-63579, FLOW-64101) Fixes and improvements will be made available in upcoming SDK releases as usual. Thanks for additional reporting, it helps us keep priority on the issue.