polarofficial/polar-ble-sdk #575: PS-FTP MTU not enabled

janbarari, 2025-03-19:
**Platform on which you observed the bug:** 
- [x] Android 

**Device on which you observed the bug:**
- [x] Polar 360

**Describe the bug**
`com.polar.androidcommunications.api.ble.exceptions.BleCharacteristicNotificationNotEnabled: PS-FTP MTU not enabled`
Can you help me resolving the exception? When this happens and how to avoid it. I can see sometimes suddenly this exception thrown in the middle of an action

**How to Reproduce**
SDK: 5.16.0
Firmware: 1.2.11
Android API: 35

palmqvisti, 2025-05-16:
@janbarari do you still have this exception? Which api method call(s) is this related to?

palmqvisti, 2025-06-09:
@janbarari are you still experiencing this issue and can you provide additional details or can this issue be closed?

janbarari, 2025-06-10:
Haven't occurred in latest updates.

closed 2025-06-10 by janbarari