polarofficial/polar-ble-sdk #579: get247HrSamples hangs on Polar360

Marcus122, 2025-03-26:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X] Polar 360
- [ ] Device is not relevant for this bug

**Describe the bug**
Hi, I'm calling the api method get247HrSamples on the Polar360 passing in yesterdays date as the from and todays date as the to ( different combinations get the same result as well ). I sometimes get a response but sometimes the method just hangs without receiving a response. When I do receive a response this normally takes about a minute..  The other methods on the activity API are much quicker to respond.. Is this correct or should the method be a lot quicker than this?

**How to Reproduce**
Call the api method get247HrSamples using the Polar360

**Expected behavior**
The function to respond a lot quicker

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

pth-pe-gh, 2025-03-28:
@Marcus122 There are differences with storage formats for different data so response times vary because of this. 247 HR and PPi data by date fetches may be noticeably slower to retrieve, and this difference is more noticeable when there is more data stored on Polar 360. Therefore, it is recommended that data that has been already read should be removed from device. For example, daily data from past dates should be removed to avoid fill up the memory and keep fetches by date more efficient. You can use `deleteStoredDeviceData` method to delete data by type; or `deleteDeviceDateFolders` method to remove all data stored by date.

Marcus122, 2025-03-31:
@pth-pe-gh Thanks the response - I'll delete some data see if I get better response times. I can see the method `deleteStoredDeviceData`  in the documentation but not `deleteDeviceDateFolders` - is it an undocumented method?

pth-pe-gh, 2025-03-31:
@Marcus122 - here is documentation to the method for [iOS](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Protocols/PolarBleApi.html#/s:11PolarBleSdk0aB3ApiP23deleteDeviceDateFolders_04fromG002toG07RxSwift17PrimitiveSequenceVyAG16CompletableTraitOs5NeverOGSS_10Foundation0G0VSgARtF) and [Android](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarBleApi.html#deleteDeviceDateFolders(String,LocalDate,LocalDate))

Marcus122, 2025-04-01:
@pth-pe-gh Thanks for the documentation. So I've cleared all the data ( freeSpace: 10223616 ) - the method is still taking 1 or event 2 minutes or even more and then returns an empty list. Is there anything else I can try to return results and with a quicker response?

If I try and use another method while waiting for the 24HrSamples response I get the error - Air packet lost! Error: 303.message - I don't know if this helps at all?

pth-pe-gh, 2025-04-01:
It you have used both methods, it sounds like there is some data still left over after those actions; also the free space you report is quite a bit less than what is available after factory defaults. You can also use factory defaults to erase store data completely, do first-time-use again and then collect new data with more free memory available, then compare the performance to verify if left over data causes the delay. 

We will check if there are issues in current data delete methods per day or type.

Can you give the firmware version you are using? Latest available is 1.2.11 for Polar 360.

pth-pe-gh, 2025-04-03:
@Marcus122 - when you used `deleteStoredDeviceData`, did you delete all available data types: SKIN_TEMP, SKIN_CONTACT_CHANGES, SLEEP_SCORE, SLEEP, SDLOGS, NIGHTLY_RECOVERY, DAILY_SUMMARY, AUTO_SAMPLE, ACTIVITY ?

Marcus122, 2025-04-03:
@pth-pe-gh Yes I did and I had called `deleteDeviceDateFolders` up to the current date. So today the method is working.. is it correct as when passing "fromDate" as yesterday and "toDate" as today only returns todays data.. to get yesterdays data I have to pass a "fromDate" as the day before yesterday?

Even after a factory reset with only 1 days of data the method is taking 20secs+ to respond, is that what you'd expect? Here's the disk space I currently have `"totalSpace": 15024128, "freeSpace": 14286848`

pth-pe-gh, 2025-04-04:
@Marcus122 - 24/7 data for 24/7 is stored timestamped on device, and the parameter to  `get247HrSamples` should be closed range from fromDate to toDate; this means you should be getting data time stamped for yesterday with any date range that includes yesterday's date. At minimum I'd expect this range needs to be from yesterday's date to yesterday's date. If you are not getting this I'd say either the device does not have the data timestamped within yesterday's date; or there could be a bug in the SDK method. If you see evidence of the latter, please file a bug.

As for retrieving the single day's 247 HR data, I'll check with our QA on typical response times for comparison.

pth-pe-gh, 2025-04-07:
@Marcus122 - I can confirm 20+ seconds response for getting 247 HR or PPi is not uncommon after a few days worth of data has been saved on device. Also, the team is working on improvements on this. However, current devices, firmware, and the method data is stored requires frequent data read + removal from device to keep things going smoothly.

Marcus122, 2025-04-08:
@pth-pe-gh Thanks for all in the info. We're just trying to sync data in a background task on ios and android but are limited to a 30 seconds window as a limitation from the OS so hopefully if we remove the data regularly we should be able to do it.

Marcus122, 2025-05-21:
@palmqvisti Yes we've seen some improvement. For 247HR we run deleteStoredDeviceData with AUTO_SAMPLE up to yesterdays date ( which can take a while ) then running 247HR for yesterday to today runs in approx 12secs. This seems to be as quick as we can get it without clearing all of todays data.

Can I just check is it just AUTO_SAMPLE that needs clearing for the 247 data? Can everything else be cleared by using deleteDeviceDateFolders?

palmqvisti, 2025-05-21:
@Marcus122, yes, AUTO_SAMPLE is the only data that deleteDeviceDateFolders does not remove.

Marcus122, 2025-05-21:
@palmqvisti One more question - when we call one of the delete methods can we start the method and not wait for the completable to finish? We're trying to clear in a background job on ios/android which is limited to 30secs - so if the process is killed does the device still carry on and finish the deletion?

palmqvisti, 2025-05-22:
@Marcus122, yes, the device continues to execute the process. But when deleting offline recordings if the recording is splitted it is deletes only one piece of the recording and the rest of the pieces remain undeleted.

palmqvisti, 2025-06-09:
@Marcus122 are you still experiencing issues or can this issue be closed?

Marcus122, 2025-06-11:
@palmqvisti All sorted - will close

closed 2025-06-11 by Marcus122