polarofficial/polar-ble-sdk #580: Synchronize two polar H10 timestamps

jukaradayi, 2025-03-28:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hello,

I am reaching out regarding the time stamps of the Polar H10. We looked at the documentation and the Github issues but could not find a solution to the below problem.

Description: We aim to use several H10 to synchronously record ECG from several individuals for a scientific study. In our project, each Polar H10 is connected through BLE to a custom program that starts the recording and saves the ECG data onto local text files. To be able to interpret the recorded data afterwards, we need to make sure that the timestamps of the different Polar H10 are perfectly aligned.

Problem: To this effect, we have tried using the setLocalTime feature of the Android application at the beginning of the recording session, after the Polar H10 are installed on individuals. The Android application responds as expected (confirming a new time has been set), and the recorded data indeed show that the time of the Polar H10 has been successfully set to the current time. However, comparing between two Polar H10 installed on the same individual, we can see that there is a lag (of 100 to 160ms) between the two recordings, hence, that timestamps of the two belts are not aligned.

Question: Do you have any idea what could cause this lag and how we could avoid it? Android API's Date() has millisecond precision so the setLocalTime feature supposedly aligns timestamps with a comparable precision. Is there any alternative solution to align the ECG recordings from different Polar H10?

Thank you in advance for your time and please let me know should you need any further information.

mattisipola, 2025-04-04:
You may also use *mBluetoothGatt.requestConnectionPriority(BluetoothGatt.CONNECTION_PRIORITY_HIGH);* before sending setLocalTime command (some seconds after priority request so that connection parameters can have time to change to faster, remember to change back to low power or disconnect). (If this is possible when using Polar BLE SDK?)

pth-pe-gh, 2025-04-07:
See https://github.com/polarofficial/polar-ble-sdk/issues/573 on a possible method of providing sync data point when aligning the data from multiple sensors. Let us know if you were able to apply something similar in your case to solve your multiple clock time difference problem.

chungv, 2025-04-14:
Thank you both for your insights ! Based on your suggestions, I could run a few tests to try synchronizing the clocks of two Polar H10 with each other. I actually see two issues that may or may not be related.

**Testing parameters**
- One individual simultaneously wears the two sensors (middle of thorax, upper abdomen). 
- The clocks of the Polar H10 sensors are set one after the other using the Polar SDK's Example App running on an Android 13 device which provides a common reference time. The distance between the Android device and the sensors does not change.
- Expected results: if the clocks of the two sensors are aligned, the timing of the recorded heartbeats should be identical or nearly identical. There might be a small difference in the timing of hearbeats because the sensors are not positioned at the exact same location, but this difference is expected to remain fixed over time.

**Issue 1: the accuracy of setLocalTime varies from a few to over hundred milliseconds**
@mattisipola  's suggestions : the same Android device was used to set the time of the two sensors, one after the other; only one Bluetooth device was connected at a given time; WiFi was deactivated; the setLocalTime function was used after the app confirmed the sensor was successfully connected and before 30 seconds had passed from connection. I could not use 'BluetoothGatt.requestConnectionPriority(BluetoothGatt.CONNECTION_PRIORITY_HIGH)' because my knowledge in app development is too limited.
Test: I have used the setLocalTime function after varying delays (from 2 to 10 seconds) after the connection event.
Results: the offset between the sensors' clocks varied in a seemingly random fashion (from 6 to 115 milliseconds), although the abovementionned conditions were always met. Replicating the exact procedure several times led to different offsets between the sensors' clocks. 
Question: Is there any other parameters that could affect the accuracy of the SetLocalTime procedure?

**Issue 2: the offset between the sensors' clocks increases over time**
Although it is expected that the clocks of two sensors may drift apart over time, I was surprised to see that the initial offset between the clocks was more than 7 times bigger than the initial offset (see Figure A) after 2 hours of recordings (see Figure B). As can be seen from Figures C and D, the offset between the sensors' clocks varies in a jiggsaw pattern and increases at sudden and regular jumps that occur about every 15 minutes or so. 
Question: Any idea where these jumps may come from and possibly how to at least mitigate their impact (e.g., reduce the frequency of these jumps)?

Thank you again for your help, any advice is much appreciated.

[image]
[image]
[image]
[image]

mattisipola, 2025-04-15:
It seems you have reached quite accurate H10 time synchronization. With standard BLE connection it is very hard to get better.

Comparing R-peaks from ECG-samples is not very accurate, because sample rate is about 130 Hz with 1% accuracy. It is about 7,7 ms between samples. Very rarely that maximum value of a sample is the real R-peak and that's why they randomly differ 0-7,7 ms.

Also I checked connection indication packets from two different Android 14 phones. Sony had 45 ms and Samsung had 30 ms. With earlier Android versions 7,5 ms was mostly used, but these longer intervals are much better for a coin-type battery (CR2025). When you send setLocalTime with default initial connection interval it may take quite long random time over the air to reach H10. If possible you can try to send SetLocalTime command exactly 7500 ms after Connected event. At that time Service Discovery has been finished and H10 should not be so busy.

If you could force CONNECTION_PRIORITY_HIGH for the time when you send setLocalTime you could achieve always 0-7,5ms accuracy. There is Android example for that in developer pages. You can also re-sync clocks with some time period.

From the picture C drifting is normal because 32 kHz crystals have 20 ppm accuracy that means clocks on both H10 may drift 20 us in a second. Because temperature should be the same for both H10 you should have a new battery on both device. Supply voltage may also affect to accurate 32768 Hz. H10 inserts UTC nano second timestamps in the front of each larger ECG notification packet. Maybe it's better to calculate actual sample rate from the time stamps. After warming up sample rate does not change much.

This about 7,7 ms jumps during drifting seems to be quite weird. Actually it is time between two samples. Can you check is there one sample missing or does H10 make some time calibration. You can try re-sync before that jump happens.

dntil, 2025-05-09:
Hi @mattisipola, can you please share a demo or some code how to `requestConnectionPriority`? I didn't see how I can get the mBluetoothGatt object through Polar api. Thank you.

mattisipola, 2025-05-13:
@dntil if you start ECG streaming, H10 will request faster parameters automatically. Usually phone changes connection interval to 30 ms. If you synchronize your SetLocalTime to incoming ECG packets just before Android ns accurate time changes seconds, then you will get better synchronizing of time to all H10 devices. You can check H10 time by reading timestamp from the last ECG packet. It's the H10 local time for the last ECG sample just before packet has been sent.

closed 2025-07-04 by rkangast