polarofficial/polar-ble-sdk #581: Inquiry Regarding Skin Temperature Data Streaming with Polar360

K-sim023, 2025-03-31:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Other
- [ ] Device is not relevant for this question

**Description:**

I am currently attempting to use Polar360 for online streaming of skin temperature data. I would like to inquire about the following points:

1. Could you please explain the difference between "Temperature" and "SKIN_Temperature" in PolarDeviceDataType?
2. Specifically, I have concerns about whether skin temperature can actually be obtained when streaming using the "Temperature" data type in Polar360. Any clarification on this matter would be greatly appreciated.

Thank you for your attention to this matter. I look forward to your response.

rkangast, 2025-03-31:
Hello,

Temperature is the temperature of a device where as skin temperature is measured from the surface of your skin. However in Currently, skin temperature data type is not used in Polar 360 device. You can use temperature device data type for skin temperature request to device.

K-sim023, 2025-03-31:
Thank you for your rapid response! It's great to know that SKIN_Temperature online streaming is also possible with the Polar360.

closed 2025-03-31 by K-sim023