polarofficial/polar-ble-sdk #582: Continuously stream HR data from Polar Vantage V3 without training running

omatt, 2025-04-02:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Other
- [ ] Device is not relevant for this question

**Description:**
Is it possible to stream HR data from the watch without running a training? As mentioned on this [ticket](https://github.com/polarofficial/polar-ble-sdk/issues/456), fetching HR data can be done when "training." I do see that the watch is continuously reading HR from the **Daily activity** view on the watch, but no HR data is being returned on the SDK.

omatt, 2025-04-03:
How can the watch device be set to an "exercise wait state"? Is it mentioned anywhere in the docs? Is it also possible to set the watch device to an exercise wait state from the mobile app using polar-ble-sdk?

rkangast, 2025-04-03:
Select "Start training" and then select some sport activity but do not start the training. We call this state "exercise wait". Unfortunately it is not possible to do this via SDK.

closed 2025-04-03 by omatt