polarofficial/polar-ble-sdk #584: Polar 360 startPpiStreaming gives error.

abbjetmus, 2025-04-07:
**Platform on which you observed the bug:** 
- [X ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Polar 360
- [ ] Device is not relevant for this bug

**Describe the bug**

I'm trying to stream PPI values on Polar 360 but get this error.
Streaming ACC values works fine.

[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command REQUEST_MEASUREMENT_START error: ERROR_ALREADY_IN_STATE, pmd cp command REQUEST_MEASUREMENT_START error: ERROR_ALREADY_IN_STATE, null, null)

**How to Reproduce**
Start streaming PPI with Polar 360

**Expected behavior**
Should stream PPI values like polar verity sense does
**Screenshots and logs**
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command REQUEST_MEASUREMENT_START error: ERROR_ALREADY_IN_STATE, pmd cp command REQUEST_MEASUREMENT_START error: ERROR_ALREADY_IN_STATE, null, null)

mikeread-dev, 2025-04-07:
I don't think it's possible to do both streaming and offline recording: 
`The online streaming and offline recording do not work same time for the same data type. For example if either accelerometer offline recording or the accelerometer online streaming is started, the attempt to start the online streaming or offline recording at this state will return the error ERROR_ALREADY_IN_STATE.`
https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/SdkOfflineRecordingExplained.md

abbjetmus, 2025-04-08:
You are right @mikeread-dev but even if I call StopOfflineRecording I still get the error trying to start streaming PPI.

rkangast, 2025-07-04:
This issue has been silent for some time now. I will close this issue now, please do reopen this if needed.

closed 2025-07-04 by rkangast