polarofficial/polar-ble-sdk #585: SDK package missing from Android release 6.0.0

JohnyGQD, 2025-04-08:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [X] Device is not relevant for this bug

**Describe the bug**
As SDK 5.17.0 does not include firmware version check (or I just wasn't able to find it), I wanted to update to 6.0.0. However, all the SDK packages went missing. After investigating the dependency, it seems that SDK package is not included in the AAR.

**How to Reproduce**
Update SDK to 6.0.0 in your Android Project.

**Expected behavior**
Access to SDK package classes such as PolarBleApi

**Screenshots and logs**
[image]
[image]

TeemuLauPolar, 2025-04-13:
FLOW-64410

JohnyGQD, 2025-04-15:
Same issue with 6.1.0

closed 2025-07-04 by rkangast