polarofficial/polar-ble-sdk #588: Error when parsing offline recording created by Polar 360

viesti, 2025-04-09:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Other (Polar 360, firmware 1.2.11, hardware revision 007111787.01)
- [ ] Device is not relevant for this bug

**Describe the bug**

I'm downloading offline recordings from Polar 360 device with `getOfflineRecord` and seeing errors after download, when the SDK is parsing the results.

The error I see is "Failed to parse data" at https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/PolarBleApiImpl.swift#L1468

With a tweak like this, I was able to get the original error message:

```
                                                  } catch {
                                                      NSLog("OfflineRecordingData parsing error: \(error)")
                                                      throw PolarErrors.polarOfflineRecordingError(description: "Failed to parse data")
                                                  }
```

and on of the errors is:

```
gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")
```
I tried another hack to grab the binary data:

```
                                                  do {
                                                      dataResult.write(to: FileManager.default.urls(for: .documentDirectory, in: .userDomainMask)[0].appendingPathComponent("data.bin"), atomically: true)
                                                      let pmdSecret = try secret.map { try PolarDataUtils.mapToPmdSecret(from: $0) }
```

Here is one of the failing files [data.zip](https://github.com/user-attachments/files/19666904/data.zip)

I did see these errors in a few files (don't have file names in the logs, but I see errors with PPI, HR, Temperature and ACC data types):

```
OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")
OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")
OfflineRecordingData parsing error: offlineRecordingErrorMetaDataParseFailed(description: "offlineRecordingErrorMetaDataParseFailed(description: \"Invalid range\")")
OfflineRecordingData parsing error: offlineRecordingErrorMetaDataParseFailed(description: "offlineRecordingErrorMetaDataParseFailed(description: \"Invalid range\")")
OfflineRecordingData parsing error: offlineRecordingErrorMetaDataParseFailed(description: "offlineRecordingErrorMetaDataParseFailed(description: \"Invalid range\")")
OfflineRecordingData parsing error: offlineRecordingErrorMetaDataParseFailed(description: "offlineRecordingErrorMetaDataParseFailed(description: \"Invalid range\")")
OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")
OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")
OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")
OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")
OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")
```

**How to Reproduce**

I'm yet not exactly sure how to reproduce this problem. I did try starting offline recording and putting the device to charger and resuming recording multiple times. Also I think I ran into disk running full (had to remove recordings to resume recording). I'll report back if I'm able to narrow the situation.

**Expected behavior**

Recordings getting created without error.

TeemuLauPolar, 2025-04-13:
FLOW-64411

viesti, 2025-04-13:
Hmm, might be related that I specified empty sensor settings when starting streams, since for temperature I seem to constantly get:

> OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")

I think HR and PPI don't support settings, but temperature should have a sample rate specified, maybe?

viesti, 2025-04-14:
Ok, I think I managed to reproduce this. I think the problem is that the API allows to start temperature offline recording without specifying any settings, which gives a sample rate of zero. Somehow data is still recorded, but when reading the data, since sample rate is zero and sample rate would be needed to calculate timestamps, the reading fails.

What I don't yet understand is that how I saw this for PPI and HR as well. The online streaming API doesn't allow to specify settings for HR and PPI streaming, but the offline recording API does allow specify settings, so there is discrepancy and I had actually specified settings to temperature online streaming, but forgot to do so for offline recording.

What settings actually are applicable per each data type? For temperature online streaming I have:

```
        let tempSettings = PolarSensorSetting([.channels: 1,
                                               .sampleRate: 1,
                                               .resolution: 32])
        api.startTemperatureStreaming(self.selectedDevice!, settings: tempSettings).subscribe(
```

But I think from these, only `sampleRate`  is used, right?

palmqvisti, 2025-04-22:
@viesti, you can use requestStreamSettings to get applicable settings for a data type.
https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Protocols/PolarOnlineStreamingApi.html

viesti, 2025-04-22:
Thank you!

I think there could still be validation of settings when starting offline recording. That said, I didn't look what happens when starting for example temperature online streaming with out-of-bounds settings.

viesti, 2025-05-02:
I'm still seeing parsing errors sometimes with HR data too:

```
 {:path "/U/0/20250421/R/210425/TEMP0.REC",
  :date #inst "2025-04-21T18:04:25.000-00:00",
  :type "temperature",
  :size 50},
 :error
 #object[CodedError Error: polarOfflineRecordingError(description: "Failed to parse data")]}
```

The file size seems small, maybe it is related? Also, this doesn't happen for every file, just sometimes. I haven't yet been able to pinpoint why.

For HR offline recording, I now just specify empty settings via `PolarSensorSetting([:])`. Is this correct?

l1va, 2025-05-06:
Hey, I experience the same. I think it is because the file is open and the data were not flushed to the disk, they are buffered before writing to the file. And for some time after you started the recording the file is still empty and of course failed to parse.  Try to stop recording before reading the file and it should work. Or wait for enough time when the device decides to flush data from memory to file, but in such case you still will have only part of data, not all of it till the last moment.

viesti, 2025-05-06:
Ah, I do stop offline recording first, then list files, and then download files. I think I did try adding a small wait after stopping recording to account for flushing, but after (brief) testing, such timeout didn't seem to be needed (and it also make the whole process take a bit longer).

The example file above has timestamp

```
:date #inst "2025-04-21T18:04:25.000-00:00"
```

is quite old by now, so writing to it had to have stopped a while ago.

So, something is leading to corrupt files, can't yet figure out what.

I now have two files

```
{:path "/U/0/20250421/R/210425/HR0.REC",
 :date #inst "2025-04-21T18:04:25.000-00:00"
  :size 39
  :type "hr"}

:error #object[CodedError Error: polarOfflineRecordingError(description: "Failed to parse data")]}
```

and

```
{:type "temperature"
 :path "/U/0/20250421/R/210425/TEMP0.REC"
 :size 50
 :date #inst "2025-04-21T18:04:25.000-00:00"}

:error #object[CodedError Error: polarOfflineRecordingError(description: "Failed to parse data")]}
```

sitting on my device that can't be parsed. I can dump them out if it helps to take a look at them (should maybe look at a healthy file first to spot differences :)).

rkangast, 2025-05-12:
Hi @viesti 

This error "OfflineRecordingData parsing error: gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 0 and sampleRate 0")" for temperature measurement occurs when the offline file from the device does not contain any measurement data; it only contains some metadata describing the file contents. This usually happens when you start a measurement and immediately stop it. It might also be possible to happen when disk is getting full, but I'm not 100% sure. 

Also the errors in this discussion happens for the same reason; there is no measurement data to be read.

It should be safe to omit the error.

viesti, 2025-05-12:
Thank you!

Right, and the file size probably correlates with the lack of measurements (39 for HR, 50 for temperature).

If possible, it would be nice to report an empty array of measurements in these cases, instead of an error, but it is good to hear this :)

palmqvisti, 2025-05-12:
@viesti, that is a good improvement suggestion. I will add it to our backlog and close this issue for now.

viesti, 2025-05-20:
Maybe not best to report to close issue but I think I saw a case now, where I had a single 39 byte HR recording file on the device, and even if telling device to start offline recording again, it didn't create new files.

Only after removing the failed file did new files start get created.

```
{:size 39,
  :date #inst "2025-05-07T14:24:55.000-00:00",
  :type "hr",
  :path "/U/0/20250507/R/172455/HR0.REC"}
```

I'm not totally sure if a "failed" file caused this, maybe there is some other reason.

Also, it now seems that I get constantly a "39" byte HR file, even after removing such file. A temperature file is getting created, after removing a 39 byte HR file, only same size file gets created. Temperature file does get created and when listing files, temperature file seems to grow in size, but HR file not.

So kind of seems that firmware maybe got stuck in creating offline recording data?

viesti, 2025-05-20:
Hmm, ok, now saw a HR file growing in size. Maybe I'm seeing artifacts of buffering, but it felt as is a "empty" file caused a problem for creating new files.

rkangast, 2025-05-21:
Might be the root cause, but I cannot be 100% certain either. It might be, especially for shorter measurement sessions, that the measured data does not get flushed into the file before the measurement has been stopped.

viesti, 2025-05-21:
Ah, thank you!

Is the stopOfflineRecording() synchronous from the point of view of SDK user? Meaning that does there need to be a wait time after calling stopOfflineRecording() and waiting on the completion of the observable, or is it enough to wait on the observable only?

Another question comes to mind that is there a recommended minimum time to wait between offline recording stopping? I think flushing more often leads to more wear of the internal memory.

I think we had way too low minimum synch interval (2 minutes, although synch now happened when app transitions to foreground, so there should have been usually more than 2 minutes between stopOfflineRecording() calls.).

Also, just to clarify, the 8 minutes mentioned here means flushing without user calling stopOfflineRecording(), right? Just thinking that if firmware flush can be ~8 minutes, does this also give a hint towards expected durations between flushes requested by SDK user?

rkangast, 2025-05-21:
It is an asynchronous method so you should wait for response. Please take a look at our iOS example app [here](https://github.com/polarofficial/polar-ble-sdk/blob/8aabd3afeb5ca9f45540ebc26b0d901063cfa859/examples/example-ios/iosBleSdkTestApp/iosBleSdkTestApp/PolarBleSdkManager.swift#L448-L465).

viesti, 2025-05-21:
Yup, we wait on the completion of the RxSwift observable, so I think we should be covered.

I wanted to just ask that does the stopOfflineRecording() operation also block for the flush on the firmware side, or does it complete immediately. It sounds like it does block on flush so waiting on the completion should then be enough on the client SDK side :)

rkangast, 2025-05-21:
Yes. The method waits for response from the device.