polarofficial/polar-ble-sdk #589: Version 6.1.0 is not implement correctly.

janbarari, 2025-04-11:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

Have you tried adding the 6.1.0 dependency in a project? It seems `com.polar.sdk` package and all the files included are not resolvable

I used `com.github.polarofficial:polar-ble-sdk:6.1.0` in my project and I am sure the jitpack maven also added

TeemuLauPolar, 2025-04-13:
FLOW-64410

paulirotta, 2025-04-15:
Same problem here, no work-around found. Updating java to 21 didn't help. It seems 2.0.0 is no longer available.

TeemuLauPolar, 2025-04-16:
Thanks for the report!
Fixed in version 6.1.1. Let us know if you still encounter issues after updating.

paulirotta, 2025-04-24:
Good fix. Thanks!

closed 2025-04-24 by janbarari