polarofficial/polar-ble-sdk #590: Some questions to get started with the Activity API

abbjetmus, 2025-04-19:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X] Polar 360
- [ ] Device is not relevant for this question

**Description:**

Hi!

I using a Flutter wrapper on the Polar SDK and I would like to get Step Counts and maybe additional information using Polar 360. Currently we only store PPI data offline. To get step counts what is the requirements, do I have to put the sensor in a 24/7 Activity Mode first? Since I need to write wrapper code for the functionality that doesn't exist in the Flutter wrapper. 

Or can I just wait for the feature to be ready: Polar activity API. Requires feature FEATURE_POLAR_ACTIVITY_DATA.

[image]

In this state with this api does the sensor store all this information and I can call with whenever I want.

Also I see in your app that you have something that is called "Inactivty Alert". How do you achieve that from you app. Is by sending data in the background with an interval?

[https://support.polar.com/e_manuals/A360/Polar_A360_user_manual_English/Content/Activity_Tracking.htm](https://support.polar.com/e_manuals/A360/Polar_A360_user_manual_English/Content/Activity_Tracking.htm)

rkangast, 2025-04-22:
Hi,

Steps are automatically collected by the device as part of the daily activity data collection. There is no need to set sensor to any specific state. You can read the steps using the [activity API ](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarActivityApi.html). Steps data is stored at the device and you can call the API whenever you want to.

Inactivity alert has not been not implemented in SDK and actually that Polar **A**360 device mentioned in our support.polar.com page is different from [Polar 360 device](https://support.polar.com/fi/polar-360/).

Hope this helps :-)

abbjetmus, 2025-04-22:
Thanks @rkangast! I don't know why though calling getSteps with two different days causes the crash above. Maybe only getSteps for same day are allowed?

rkangast, 2025-04-23:
Which version of the SDK do you use? We did some improvements on this API in version 5.16.0. Does this error still occur with version 5.16.0 (or higher)?

abbjetmus, 2025-04-23:
I upgraded to latest version and still get errors.

These dates doesn't work causes the app to crash with the errors earlier:

[image]

[image]

These dates work:

[image]

[image]

I'm not sure why that is if fromDate can't start from a date when there is not data/steps collected.

rkangast, 2025-04-23:
Sounds like a bug in SDK, we will investigate this.

abbjetmus, 2025-05-05:
@rkangast we want to ship this feature next week would be great if it is fixed this week? As far as I can tell it's only `getSteps()` that doesn't work.

rkangast, 2025-05-05:
Hi @abbjetmus,

Our next release is being released today. Fix for this issue is not included in that release, but will be in the next subsequent release.

abbjetmus, 2025-05-05:
We relly need this to be fix on latest 12 May morning when we new onboardings.

rkangast, 2025-05-06:
Hi @abbjetmus 

I took a look at this problem. Unfortunately I cannot reproduce this problem. The device will report if the requested file cannot be found with error response "Request failed:  Error: 103". That is just fine; the file does not exist on the device.

That error gets caught [here](https://github.com/polarofficial/polar-ble-sdk/blob/6.2.0/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/utils/PolarActivityUtils.kt#L86) and it is treated as a success (missing file).

closed 2025-07-04 by rkangast