polarofficial/polar-ble-sdk #592: checkFirmwareUpdate not available on Android

JohnyGQD, 2025-04-22:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [X] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
While iOS version of the SDK can check for firmware update before actually running it, there is no such thing (at least not public) in Android SDK. The `checkFirmwareUpdate` method is undefined in `PolarBleApi` and `FirmwareUpdateApi` expects `BDBleApiImpl`'s internal fields and methods to be called.

**How to Reproduce**
Try to check for latest firmware within Android SDK without actually updating it right away.

**Expected behavior**
I expect the SDK to be able to check for any new firmware without actually performing an update immediately on Android just as it does on iOS.

rkangast, 2025-04-22:
Hi,

There is no suitable API in SDK available. Instead, you could use this [firmware update api](https://github.com/polarofficial/polar-ble-sdk/blob/6.1.1/sources/Android/android-communications/library/src/main/java/com/polar/androidcommunications/http/fwu/FirmwareUpdateApi.kt) from android-communications package.

JohnyGQD, 2025-04-29:
@rkangast That's definitely the path I wanted to take, but building the `FirmwareUpdateRequest` seems rather involved and seems to require me to reimplement a whole lot of `BDBleApiImpl` (as I hinted in the original bug: _"FirmwareUpdateApi expects BDBleApiImpl's internal fields and methods to be called."_) which doesn't seem right, especially when iOS SDK only requires the `deviceId`. But maybe I understood it wrong. If it is ready as it is, then I would really appreciate some documentation. Thank you very much!

pth-pe-gh, 2025-04-30:
SDK API addition for this is in Polar development backlog for Android SDK (FLOW-59825)

palmqvisti, 2025-10-07:
API for checking for firmware update on Android is released in SDK 6.5.0

closed 2025-10-07 by palmqvisti