polarofficial/polar-ble-sdk #594: Sleep data on Polar 360

mikeread-dev, 2025-04-24:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x ] Flutter wrapper (Android)
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ x] Polar 360
- [ ] Device is not relevant for this question

I am getting sporadic results retrieving sleep data from a Polar 360.  Some days it appears after the first sync in the morning, sometimes it takes a few syncs, sometimes it never gets returned for a day.   A couple of questions about sleep data on the 360, so I can figure out it it's something I am doing wrong:

1. When is the sleep data calculated on the device?
2. What would cause the device not to calculate sleep data?
3. Is there anything I can do to force the sleep data calculation to trigger?

If I debug at the earliest point in the api wrapper code I get a blank record to today:

```
I/System.out(13054): [PolarPlugin] Sleep record #7:
I/System.out(13054): [PolarPlugin]   - Date: 2025-04-24
I/System.out(13054): [PolarPlugin]   - Has result: true
I/System.out(13054): [PolarPlugin]   - Sleep start: null
I/System.out(13054): [PolarPlugin]   - Sleep end: null
I/System.out(13054): [PolarPlugin]   - Sleep cycles count: 0
I/System.out(13054): [PolarPlugin]   - Sleep phases count: 0
```

viesti, 2025-04-25:
I've noticed that you need to call `stopSleepRecording()` to force get the sleep analysis sometimes. Got this info when asking from Pertti Almqvist:

> There may be up to 90 min delay from when you actually wake up to the automatic stop of sleep recording. That is to prevent sleep being stopped due to interruptions in the middle of the night. SDK has method to manually stopping the sleep so a stop sleep functionality can be implemented to app. After stopping the sleep (manually or automatically) the device calculates the metricst and the results should be available in a few seconds. Same applies to Nightly Recharge, it is also triggered by sleep stop.

mikeread-dev, 2025-04-25:
Thank you @viesti I'll give that a try.  Maybe going from bed to desk with little activity causes it to think I'm still sleeping!

closed 2025-04-28 by mikeread-dev