polarofficial/polar-ble-sdk #595: H10 disconnecting for a couple hours a night

technicaltitch, 2025-04-28:
***Platform your question concerns:*** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I am streaming ECG and RR to my Android device overnight.

I am using a brand new H10 battery, the phone is on my mattress, I am skinny, but my body seems to block re-connection.

However I can put the H10 inside a metal box inside my fridge, and it stays connected and successfully streams (noisily). I can take the lift 9 floors away and come back again and the connection is live and streaming continues. My skinny chest seems unlikely to block the BLE signal as much as a metal lift and 9 floors of concrete.

The Android service monitors the `api.startEcgStreaming` `subscribe` `onNext` consumer, and if it isn't called for 30 seconds, disposes the streams and calls `api.disconnectFromDevice`. The api `bleSdkFeatureReady` callback resumes the streams. The service is running fine and reconnects often succeed, but every night I miss an hour or two. (I also tried a `BroadcastReceiver` to detect `ACL_DISCONNECT` then dispose streams and `api.disconnectFromDevice`, but I don't _think_ this revealed any I wasn't already detecting via the API or `startEcgStreaming` consumer.)

Before scrapping the last few months of work based on the ECG approach, is there anything I can improve about my connection resumption? I am hoping I am overlooking something, as I can only replicate this issue by spending 8 hours in bed!

pth-pe-gh, 2025-04-28:
Hi @technicaltitch - in order to understand a little better, a few questions to you: 
- You are using the latest SDK and H10 firmware?
- Do you see any common pattern in the disconnections with your H10?
 For example, is it roughly total 2 hrs per night, but divided into a number of shorter disconnects. If so, is there a typical number of disconnects / and or periods of missed online stream? Do the connections drop around the same times or are there a lot of variations night by night?
- Do you have any info about your movements and posture  changes while you sleep that could have any effect on the connections? 
l'll forward the question on best way to resuming connections to SDK Android developers in our team so they can advise.

technicaltitch, 2025-04-28:
Thank you very much indeed.

Yes latest H10 firmware. Good question sorry - I will try 6.1.1 tonight. I had been using these (copied from the sample Android project):

```
implementation 'com.github.polarofficial:polar-ble-sdk:5.5.0'
implementation 'io.reactivex.rxjava3:rxjava:3.1.6'
implementation 'io.reactivex.rxjava3:rxandroid:3.0.2'
```

It tends to be a solid 1 to 2 hours near the end of the night but recording does resume before I wake up. There was one occasion when I think my posture was turned away from the phone during the disconnection, but I've very little to go on as each night I learn a little more about what to log to file.

Thanks again, I'll report back in ~10 hours.

technicaltitch, 2025-04-29:
I upgraded to 6.1.1, added an `api.cleanup()` before disconnecting, and reduced MTU to 70, and got only 17 minutes disconnected last night. Today I will remove the broadcast receiver as the `PolarBleSdk` always detected disconnects just as promptly. Current feeling is 17 minutes is still too much though, as I'll certainly have users with more electromagnetically insulating chests than me.

pth-pe-gh, 2025-04-29:
If MTU reduction had effect, it is worth experimenting more with the value to see if further improvement can be gained. Also, you mention placing phone on the floor, so have you tried to place the phone higher and/or closer to you and H10. You could also investigate how posture affects the connection and receiving data may be to try to also collect acceleration values and  try to confirm see if there are movements prior to losing connection, and maybe try to recognise what those are. Naturally you could also try to video record your posture changes overnight to confirm that sensor vs body vs phone positioning.  Insight what type of advice to give your users on the setup required for more reliable data collection.  All this is of course time consuming but quite specific to your use case of using SDK and H10, so hopefully your efforts pay off and help you improve the data collection successfully for your users. I've also send queries to Polar research team in hopes they have more suggestions to share on this type of measurement. We do appreciate your sharing your own findings on this.

To conclude, at the moment your issue seems HW (H10 + chest strap) and environmental conditions related, rather than something related to SDK functionality; apart from the fact that H10+SDK only supports online streaming for ECG. For HR and RR you could try to use exercise recording API for overnight recording and fetch the results from H10  in the morning.

pth-pe-gh, 2025-04-29:
@technicaltitch I received some information that may be helpful to you in this case:
1) Body obstruction between H10 and Polar sports watch in certain sports has been known to block signal; so this seems to align with your experience, albeit with phone instead of wrist unit. The user switched sensor to opposite side of body, where also his wrist was located during the exercise. But maybe experimenting with strap/sensor position could be helpful in your case.
2) If conductivity of skin + strap is changing (dry conditions), conductive gel applied to strap contacts has been used. You may also try this approach to ensure sensing signal is not dropping because of skin contact issues.

closed 2025-04-29 by technicaltitch