polarofficial/polar-ble-sdk #596: Spelling mistake in PolarUserDeviceSettings.DeviceLocation enum

steve2116, 2025-05-06:
**Platform on which you observed the bug:** 
- [ ] Android 
- [X] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [X] Device is not relevant for this bug

**Describe the bug**
There is a spelling mistake on line 25 of `PolarBleSdk/sdk/api/model/PolarUserDeviceSettings.swift`
```
        case UPPER_ARM_RIGHTs
```

**How to Reproduce**

**Expected behavior**
The enum should be `UPPER_ARM_RIGHT` to match with the other enums

**Screenshots and logs**

samulimaa, 2025-05-07:
Hi, yes it's a spelling mistake. I have created internal ticket about this. (FLOW-64860)

palmqvisti, 2025-10-07:
Spelling mistake is fixed in SDK 6.3.0.

closed 2025-10-07 by palmqvisti