polarofficial/polar-ble-sdk #598: Polar 360 gets stuck in isFtuDone call

viesti, 2025-05-07:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**

6.1.0

**Customer app url**

Not yet released

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**

When connecting to device, we currently wait for `bleSdkFeatureReady()` to be called with `feature_polar_activity_data` to detect that device is ready for use.

On receiving this event, we call `isFtuDone()` but it seems that sometimes the device doesn't respond to this call.

I just now had a case where this call started to pend and didn't succeed. After killing the application process a few times, I resorted to pressing the reset pin on the device.

This helped, the next time around the `isFtuDone()` call didn't block.

**How to Reproduce**

I'm not quite sure how to reproduce. I did do factory reset to instruct a colleague how to perform the procedure, and after a while, this problem staretd to appear.

**Expected behavior**

`isFtuDone()` should complete relatively quickly. I did see in logs a "[BLE] readResponse started with timeout: 90.0 seconds", didn't get around waiting that time. Usually `isFtuDone()` call is quite fast.

**Screenshots and logs**

Check if FTU has been done to device E4937920
[BLE] PS-FTP new notification operation started for ID: 8
[BLE] Notification packets counter reset
[BLE] Transmitting notification packet, size: 2 bytes, for notification ID: 8
H2D send  HEX 02 08
[BLE] waitPacketsWritten started with count: 1, timeout: 90.0 seconds
[BLE] Current written: 0, Target count: 1
[BLE] serviceDataWritten called with characteristic: FB005C53-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0
[BLE] Incrementing notificationPacketsWritten for PSFTP_H2D_NOTIFICATION_CHARACTERISTIC
[BLE] All packets written successfully, resetting written count
[BLE] PS-FTP notification sent successfully for ID: 8
[BLE] PS-FTP Notification send operation DISPOSED for ID: 8
[BLE] Closing totalStream in PS-FTP notification operation for ID: 8
[BLE] PS-FTP new request operation
[BLE] Transmitting MTU packet, size: 22 bytes
[BLE] commandState state:IDLE action: ADMIN_STOP_SCAN
[BLE] commandState state:IDLE action: EXIT
[BLE] commandState state:STOPPED action: ENTRY
MTU send  HEX 02 13 00 08 00 12 0F 2F 55 2F 30 2F 55 53 45 52 49 44 2E 42 50 42
[BLE] waitPacketsWritten started with count: 1, timeout: 90.0 seconds
[BLE] Current written: 0, Target count: 1
[BLE] serviceDataWritten called with characteristic: FB005C51-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0
[BLE] Incrementing packetsWritten for PSFTP_MTU_CHARACTERISTIC
[BLE] All packets written successfully, resetting written count
[BLE] readResponse started with timeout: 90.0 seconds
[BLE] Device is connected, attempting to poll input queue

samulimaa, 2025-05-09:
Hi, thanks for reporting this. I have created internal ticket for further investigation. (FLOW-64899)

rkangast, 2025-06-03:
I tried to reproduce this but could not repeat the issue.

So was it so that in the situation this issue occurred the BT pairing may not have been completed before calling isFtuDone()? Your our assumption is ok, calling the isFtuDone after the callback is should be fine, BT session has been established at that point and FTP service is available, isFtuDone() is using FTP over BT connection.

viesti, 2025-08-29:
Odd, seems this appeared again. Was able to recover by pushing the reset PIN.

This time I had SDK 6.5.0 and firmware 3.0.16. 

Now it seemed that device disconnected in the middle of trying to read FTU.

The device had 14% battery, so should be enough still.

Anyways, pressing reset PIN helped, problem might not be FTU related, but something else.

Here are some logs anyway :)

```
default	23:36:27.665553+0300	EndorApp	Feature feature_polar_activity_data ready
default	23:36:27.665707+0300	EndorApp	Received last feature: feature_polar_activity_data
default	23:36:27.666467+0300	EndorApp	PolarBleApiLogger: device setup completed
default	23:36:27.666851+0300	EndorApp	PolarBleApiLogger: Check if FTU has been done to device E4937920
default	23:36:27.666944+0300	EndorApp	PolarBleApiLogger: [BLE] processServiceData called with characteristic: FB005C52-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0, data size: 2 bytes
default	23:36:27.667337+0300	EndorApp	PolarBleApiLogger: Processing PSFTP_D2H_NOTIFICATION_CHARACTERISTIC, data:  HEX 02 07
default	23:36:27.667485+0300	EndorApp	PolarBleApiLogger: [BLE] PS-FTP new request operation
default	23:36:27.667769+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:IDLE action: ADMIN_STOP_SCAN
default	23:36:27.667860+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:IDLE action: EXIT
default	23:36:27.668031+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:STOPPED action: ENTRY
default	23:36:27.668509+0300	EndorApp	PolarBleApiLogger: [BLE] Transmitting MTU packet, size: 22 bytes
default	23:36:27.668766+0300	EndorApp	PolarBleApiLogger: MTU send  HEX 02 13 00 08 00 12 0F 2F 55 2F 30 2F 55 53 45 52 49 44 2E 42 50 42
default	23:36:27.669111+0300	EndorApp	PolarBleApiLogger: [BLE] waitPacketsWritten started with count: 1, timeout: 90.0 seconds
default	23:36:27.669358+0300	EndorApp	PolarBleApiLogger: [BLE] Current written: 0, Target count: 1
default	23:36:30.723576+0300	EndorApp	PolarBleApiLogger: [BLE] serviceDataWritten called with characteristic: FB005C51-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0
default	23:36:30.723968+0300	EndorApp	PolarBleApiLogger: [BLE] Incrementing packetsWritten for PSFTP_MTU_CHARACTERISTIC
default	23:36:30.724219+0300	EndorApp	PolarBleApiLogger: [BLE] All packets written successfully, resetting written count
default	23:36:30.724498+0300	EndorApp	PolarBleApiLogger: [BLE] readResponse started with timeout: 90.0 seconds
default	23:36:30.725033+0300	EndorApp	PolarBleApiLogger: [BLE] Device is connected, attempting to poll input queue
default	23:36:55.494455+0300	EndorApp	PolarBleApiLogger: [BLE] GATT Base request notification disable for chr: 2A37
default	23:36:55.494633+0300	EndorApp	PolarBleApiLogger: [BLE] send next att notify: <CBCharacteristic: 0x1463abae0, UUID = 2A37, properties = 0x10, value = {length = 2, bytes = 0x0000}, notifying = YES>
default	23:36:55.533179+0300	EndorApp	PolarBleApiLogger: [BLE] GATT Base notifyDescriptorWritten for chr: 2A37 enabled: false err 0
default	23:36:55.533334+0300	EndorApp	PolarBleApiLogger: [BLE] default session state update from:  sessionOpen  to:  sessionClosing
default	23:36:55.533559+0300	EndorApp	PolarBleApiLogger: [BLE] Completed session close tear down for session Polar 360 E4937920
default	23:36:55.533803+0300	EndorApp	PolarBleApiLogger: [BLE] didDisconnectPeripheral:  <CBPeripheral: 0x1461a24c0, identifier = E50FA7A9-1DDE-45B6-A582-06A0FA043B46, name = Polar 360 E4937920, mtu = 23, state = disconnected>
default	23:36:55.533902+0300	EndorApp	PolarBleApiLogger: [BLE] default session state update from:  sessionClosing  to:  sessionClosed
default	23:36:55.534081+0300	EndorApp	PolarBleApiLogger: [BLE] PS-FTP reset
default	23:36:55.534320+0300	EndorApp	PolarBleApiLogger: [BLE][ERROR] Polling input queue failed: Toimintoa ei voitu suorittaa. (Virhe 500 kohteessa OperationCanceled.)
default	23:36:55.534348+0300	EndorApp	PolarBleApiLogger: [BLE][ERROR] PS-FTP request interrupted error: gattDisconnected
default	23:36:55.534370+0300	EndorApp	PolarBleApiLogger: [BLE][ERROR] Device disconnected during PS-FTP request
default	23:36:55.534522+0300	EndorApp	PolarBleApiLogger: [BLE] PS-FTP request operation DISPOSED
default	23:36:55.534543+0300	EndorApp	PolarBleApiLogger: [BLE] Closing totalStream in PS-FTP request
default	23:36:55.534564+0300	EndorApp	PolarBleApiLogger: gattDisconnected
default	23:36:55.534586+0300	EndorApp	PolarBleApiLogger: gattDisconnected
default	23:36:55.534636+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:STOPPED action: ADMIN_START_SCAN
default	23:36:55.534681+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:STOPPED action: EXIT
default	23:36:55.534738+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:IDLE action: ENTRY

```

viesti, 2025-08-30:
Hmm, had this happen again.

I was away from phone, and when coming back, could not connect to the device.

From logs, it actually seems that device does respond to isFtuDone() with false, and then FTU is attempted, but the devices gets stuck.

At the end of the logs I pushed the reset pin and connected again, and then was able to connect.

```
default	13:45:53.598406+0300	EndorApp	Feature feature_polar_offline_recording ready
default	13:45:53.598787+0300	EndorApp	Feature feature_polar_device_time_setup ready
default	13:45:53.599314+0300	EndorApp	Feature feature_polar_sdk_mode ready
default	13:45:53.600008+0300	EndorApp	Feature feature_polar_led_animation ready
default	13:45:53.600294+0300	EndorApp	Feature feature_polar_firmware_update ready
default	13:45:53.600532+0300	EndorApp	Feature feature_polar_activity_data ready
default	13:45:53.600666+0300	EndorApp	Received last feature: feature_polar_activity_data
default	13:45:53.600931+0300	EndorApp	PolarBleApiLogger: device setup completed
default	13:45:53.601602+0300	EndorApp	PolarBleApiLogger: Check if FTU has been done to device E4937920
default	13:45:53.601681+0300	EndorApp	PolarBleApiLogger: [BLE] processServiceData called with characteristic: FB005C52-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0, data size: 2 bytes
default	13:45:53.601936+0300	EndorApp	PolarBleApiLogger: Processing PSFTP_D2H_NOTIFICATION_CHARACTERISTIC, data:  HEX 02 07
default	13:45:53.602339+0300	EndorApp	PolarBleApiLogger: [BLE] PS-FTP new request operation
default	13:45:53.602401+0300	EndorApp	PolarBleApiLogger: [BLE] Transmitting MTU packet, size: 22 bytes
default	13:45:53.602550+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:IDLE action: ADMIN_STOP_SCAN
default	13:45:53.602792+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:IDLE action: EXIT
default	13:45:53.603039+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:STOPPED action: ENTRY
default	13:45:53.603456+0300	EndorApp	PolarBleApiLogger: MTU send  HEX 02 13 00 08 00 12 0F 2F 55 2F 30 2F 55 53 45 52 49 44 2E 42 50 42
default	13:45:53.603883+0300	EndorApp	PolarBleApiLogger: [BLE] waitPacketsWritten started with count: 1, timeout: 90.0 seconds
default	13:45:53.604080+0300	EndorApp	PolarBleApiLogger: [BLE] Current written: 0, Target count: 1
default	13:45:53.778283+0300	EndorApp	PolarBleApiLogger: [BLE] processServiceData called with characteristic: FB005C51-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0, data size: 3 bytes
default	13:45:53.779002+0300	EndorApp	PolarBleApiLogger: Processing PSFTP_MTU_CHARACTERISTIC  HEX 00 00 00
default	13:45:56.662127+0300	EndorApp	PolarBleApiLogger: [BLE] serviceDataWritten called with characteristic: FB005C51-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0
default	13:45:56.662322+0300	EndorApp	Perform FTU
default	13:45:56.665232+0300	EndorApp	PolarBleApiLogger: [BLE] Incrementing packetsWritten for PSFTP_MTU_CHARACTERISTIC
default	13:45:56.666469+0300	EndorApp	PolarBleApiLogger: [BLE] All packets written successfully, resetting written count
default	13:45:56.666629+0300	EndorApp	PolarBleApiLogger: [BLE] readResponse started with timeout: 90.0 seconds
default	13:45:56.666821+0300	EndorApp	PolarBleApiLogger: [BLE] Device is connected, attempting to poll input queue
default	13:45:56.667037+0300	EndorApp	PolarBleApiLogger: [BLE] Polled packet with size: 3 bytes, error code: 0
default	13:45:56.667706+0300	EndorApp	PolarBleApiLogger: [BLE] Processed RFC76 message frame, sequence number: 0, status: 0
default	13:45:56.667880+0300	EndorApp	PolarBleApiLogger: [BLE] RFC76 message response received with error code: Optional(0)
default	13:45:56.668026+0300	EndorApp	PolarBleApiLogger: [BLE] PS-FTP request operation completed successfully
default	13:45:56.668239+0300	EndorApp	PolarBleApiLogger: [BLE] PS-FTP request operation DISPOSED
default	13:45:56.668689+0300	EndorApp	PolarBleApiLogger: [BLE] Closing totalStream in PS-FTP request
default	13:45:56.670360+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:STOPPED action: ADMIN_START_SCAN
default	13:45:56.670649+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:STOPPED action: EXIT
default	13:45:56.671005+0300	EndorApp	PolarBleApiLogger: [BLE] commandState state:IDLE action: ENTRY
default	13:45:56.671108+0300	EndorApp	PolarBleApiLogger: [BLE] PS-FTP new query operation started for ID: 13
default	13:45:56.671387+0300	EndorApp	PolarBleApiLogger: [BLE] Cleared MTU input queue
default	13:45:56.671591+0300	EndorApp	PolarBleApiLogger: [BLE] Transmitting MTU packet, size: 3 bytes, for query ID: 13
default	13:45:56.676521+0300	EndorApp	PolarBleApiLogger: MTU send  HEX 02 0D 80
default	13:45:56.676667+0300	EndorApp	PolarBleApiLogger: [BLE] waitPacketsWritten started with count: 1, timeout: 90.0 seconds
default	13:45:56.676814+0300	EndorApp	PolarBleApiLogger: [BLE] Current written: 0, Target count: 1
default	13:45:56.868412+0300	EndorApp	PolarBleApiLogger: [BLE] processServiceData called with characteristic: FB005C51-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0, data size: 34 bytes
default	13:45:56.868809+0300	EndorApp	PolarBleApiLogger: Processing PSFTP_MTU_CHARACTERISTIC  HEX 02 08 FF FF FF FF FF FF FF FF FF 01 A2 06 13 0A 07 08 E9 0F 10 08 18 1D 12 06 08 17 10 1F 18 13 18 01
default	13:45:56.869395+0300	EndorApp	PolarBleApiLogger: [BLE] serviceDataWritten called with characteristic: FB005C51-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0
default	13:45:56.870135+0300	EndorApp	PolarBleApiLogger: [BLE] Incrementing packetsWritten for PSFTP_MTU_CHARACTERISTIC
[... cut ...]
default	13:46:52.491369+0300	EndorApp	PolarBleApiLogger: Processing PSFTP_MTU_CHARACTERISTIC  HEX 02 08 FF FF FF FF FF FF FF FF FF 01 A2 06 13 0A 07 08 E9 0F 10 08 18 1D 12 06 08 17 10 1F 18 13 18 01
default	13:46:52.491513+0300	EndorApp	PolarBleApiLogger: [BLE] serviceDataWritten called with characteristic: FB005C51-02E7-F387-1CAD-8ACD2D8DF0C8, error: 0
default	13:46:52.491619+0300	EndorApp	FTU not needed
default	13:46:52.491760+0300	EndorApp	PolarBleApiLogger: [BLE] Incrementing packetsWritten for PSFTP_MTU_CHARACTERISTIC
default	13:46:52.491884+0300	EndorApp	Resuming connectionContinuation when device ready
default	13:46:52.491955+0300	EndorApp	Device ready E4937920
```

viesti, 2025-08-31:
Remove the other copy of the app, but ran into same problem again. Have eery suspect that this is more frequent with 3.0.16 firmware.