polarofficial/polar-ble-sdk #601: Offline Accelerometer Recording Causes Device Disconnect/Reboot

l1va, 2025-05-10:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
5.16

**Customer app url**
https://apps.apple.com/at/app/thalos-ai-fitness-coach/id6744898965 

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
When I start an offline recording for the accelerometer, the device disconnects within 2 seconds and then reconnects. It appears to be more than a simple disconnection—possibly a reboot—because if other recordings (e.g., heart rate, temperature) were already active, they restart and generate new files.
By contrast, offline recordings for heart rate, PPI, and temperature work smoothly.

**How to Reproduce**
Just start ACC offline recording

**Expected behavior**
It works like hr, ppi and temp 

**Screenshots and logs**

TeemuLauPolar, 2025-05-16:
Hi,
Thank you for the bug report. We have been developing the SDK, have you already tried the latest version? Do the issues still occur with it? Based on your description, it seems that the problem may have already been fixed earlier. #355

l1va, 2025-05-17:
Hi, 
I have not tested with the latest one but 5.16 definitely newer than the fix for bug 355.

l1va, 2025-05-21:
@TeemuLauPolar  I have just checked - doesn't work with 6.3.

l1va, 2025-05-28:
@palmqvisti yes, I have done. As I wrote hr, temp and ppi work well.

l1va, 2025-07-16:
As I wrote in #652 with 6.4.0 and firmware 2.0.8 it is possible to start offlineRecording( Acc ), but it is crashing on another stage now.