polarofficial/polar-ble-sdk #602: Device Now Fails to Connect After Repeated Disconnect/Connect Cycle

l1va, 2025-05-10:
**Platform on which you observed the bug:** 
- [x] iOS

**SDK version:**
5.16

**Customer app url**
https://apps.apple.com/at/app/thalos-ai-fitness-coach/id6744898965

**Device on which you observed the bug:**
- [x] Polar 360

**Describe the bug**
One test device seems to have entered a bad state, likely after being left in a loop of connect/disconnect attempts (due to the previous accelerometer issue). It now remains permanently in the “connecting” state, even after a hard reset.
Could this be caused by storage overflow or another issue?
Is there another way to recover or reset the device fully?

**How to Reproduce**
Start automatically ACC offline recordings on "feature is ready" event and after some cycles it stops auto connect and stay in connecting state forever.   

**Expected behavior**
The device works correctly ( at least hard reset should help, the device should not die forever)

**Screenshots and logs**
No useful logs

TeemuLauPolar, 2025-05-16:
Hi,
There might be a pairing issue. Have you tried using a newer version of the SDK to see if the problem persists with that as well?

l1va, 2025-05-28:
The SDK upgrade didn't help. 

It seems that the upgrade of the firmware with the polar flow app helped.

closed 2025-05-28 by l1va