polarofficial/polar-ble-sdk #603: Clarification on Data Collection Modes and Offline Recordings

l1va, 2025-05-10:
***Platform your question concerns:*** 
- [x] Platform is not relevant for this question

**SDK version:**
5.16

**Customer app url**
https://apps.apple.com/at/app/thalos-ai-fitness-coach/id6744898965

***Device:***
- [x] Polar 360

**Description:**
I aim to collect as much raw sensor data as possible. 
I understand the Polar 360 provides data only for heart rate, PPI, temperature, and accelerometer.
I have several questions:

1. As I understand the modes are: 
Normal Online: HR (~1 Hz), PPI (1 Hz), Temp (1 Hz), ACC (50 Hz, 16-bit, ±8g)
Normal Offline: HR (~4 Hz), PPI (1 Hz), Temp (1 Hz), ACC (50 Hz, 16-bit, ±8g)
SDK Mode Online: No HR/PPI; Temp (1/2/4 Hz, 32-bit), ACC (12/25/50/100/200/400 Hz, ±2/4/8/16g, 16-bit)
SDK Mode Offline: No HR/PPI; Temp (1/2/4 Hz, 32-bit), ACC (12/26/52 Hz, ±2/4/8/16g, 16-bit)

At first am I correct in understanding? 
Is my strategy to collect normal offline data optimal? I want to collect data from four sensors 24h/day. Online means when the app is open, correct? - then online doesn't work for me.  
Is there a way to combine modes or configure the device to record both raw SDK accelerometer data and standard HR/PPI/Temp simultaneously—online or offline?

2. Heart rate records lack timestamps. Is it expected behaviour? 
For the others I see numbers like 800130377597869824 .
Is it correct that it is a 100-nanosecond ts since Jan 1, 1601 UTC, like Windows FILETIME? 

3. I found that the frequency for offline recording of heart rate is not constant, is it constant for other recordings as stated in the docs? On what does it depend?   

Thank you for the answers!

TeemuLauPolar, 2025-05-16:
Hello, 
Thank you for your question. You can find more information at:
https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/SdkModeExplained.md
https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/products/Polar360.md
https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/SdkOfflineRecordingExplained.md
https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/TimeSystemExplained.md
How to get Timestamp with heart rate value #465

l1va, 2025-05-22:
Hello, 
thank you @TeemuLauPolar 
thank you @jimmyzumthurm , I found the problem in my code regarding the frequency of offline HR. Now it seems correct.
Closing.

closed 2025-05-22 by l1va