polarofficial/polar-ble-sdk #604: Can't fetch offline recordings after 500kb in size

l1va, 2025-05-10:
**Platform on which you observed the bug:** 
- [x] iOS

**SDK version:**
5.16

**Customer app url**
https://apps.apple.com/at/app/thalos-ai-fitness-coach/id6744898965

**Device on which you observed the bug:**
- [x] Polar 360

**Describe the bug**
Start offline recording for temp or ppi for a long time. 
I have seen two behaviours (I don't know why the difference exists). See screenshots.
In the first case I see a new file was created after 500kb (like partition). If I stop recording and try to fetch it - I get responseError(errorCode: 303) 
In the second case (another device) I see files multiple times (and of a bigger size). (maybe the firmware is different)
Do you have any rules regarding the size? 
But in any case - it should fetch successfully.

**How to Reproduce**
Record data for a long time to have more than 500kb, on fetch you get responseError(errorCode: 303)

**Expected behavior**
Description of what you expected to happen. 

**Screenshots and logs**
[image][image]

TeemuLauPolar, 2025-05-16:
Hi,
Thank you for the bug report. We have been developing the SDK, have you already tried the latest SDK 6.3.0 version? Do the issues still occur with it?

l1va, 2025-05-23:
Hi, 
I tried with the latest 6.3 version, I can fetch now, but I still have a question regarding the behaviour. 

1. Is it a rule to create a separate file after the 500Kb like for the temp? Why then doesn't PPI do it? 
2. I downloaded all files ( hr, temp, ppi). hr has 123k rows, ppi has 106k rows, temp files have 244k + 3k rows. How can it be? It seems 123k rows is ok for the 1hz and ~34.3 hours, what can be a problem with ppi and temp ?    

[image]

l1va, 2025-05-23:
Addition to 2: I found that temperature files have doubled data. Like 122k rows and then again the same 122k rows. I tried to reproduce on a small file (~10 minutes) - I see 708 rows only once...