polarofficial/polar-ble-sdk #605: Connection to a movile device that has an FTU error on 360.

ymatt345, 2025-05-13:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

**SDK version:**

**Customer app url**
Link to app store / play store / your website

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [✓] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I have been using the 360 ​​for about 4 months.
I use the 360 ​​from morning to the next morning to measure,  analyze and adjust vital information.

During yesterday's measurement, the device went from DISCONNECTED, DEVICE CONNECTING, and DEVICE CONNECTED, so doFTU was executed after DEVICE CONNECTED, but an FTU error occurred.

Normally, when this process is executed, doFTU is completed without an error and the retry is successful.

However, there is a problem that once an FTU error occurs, DEVICE CONNECTED is not observed even on subsequent retries.

After factory resetting the 360, I used iOS Example app to check the symptoms and confirmed that DEVICE CONNECTED is not observed after DEVICE CONNECTING.

Another fact is that when I used iOS Example to check the symptoms on a different iOS device from the above iOS device, it worked without any problems.
General users cannot switch to another iOS device.

Furthermore, after factory resetting the 360, I used iOS Exsample to check the symptoms on the iOS device where the error occurred. Since DEVICE CONNECTED was not observed after DEVICE CONNECTING, 
I found that the 360 ​​remembers iOS devices that had an FTU error once and will not allow them to connect again.

Now, the question is, is it the right decision to not allow iOS devices that have had an FTU error once to connect?

If so, how should we deal with the fact that FTU errors occur once every few months?
POLAR may receive complaints from many users about device malfunctions in the future.

This is related to the error this time, but in my app, doFTU is executed after DEVICE CONNECTED, just like the sample app. 
Logically, if doFTU is successful once, should doFTU not be executed after DEVICE CONNECTED thereafter, or is it okay to execute it? 
I would like to know this separately from the above issue.

palmqvisti, 2025-05-19:
@ymatt345, just to note that FTU means "First Time Use" and is intended to be done only once when the device is taken into use and again after factory reset of the device. We will anyway check this issue, since FTU should not throw an exception even if it is used repeatedly.

ymatt345, 2025-05-19:
@palmqvisti 
Thank you for your comment.
I fixed the app to run the FTU only once, so that the error doesn't occur on subsequent FTUs.
However, the fact remains that the 360 ​​remembers iOS devices that have had an FTU error once and will not allow them to connect again.

rkangast, 2025-06-05:
Hello @ymatt345 

Polar 360 devices remembers the pairing information and it allows pairing only with one (same) device unless it is forced to forget the pairing. That can be done by calling doFactoryReset-api with preservePairingInformation parameter set to false. I have noticed that sometimes Bluetooth connection of a iOS device behaves unpredictably and the Bluetooth pairing needs to be removed from the mobile settings and sometimes also toggling BT on/off from mobile is needed. This usually helps in pairing the Polar 360 again. I hope this helps 🙂

closed 2025-07-04 by rkangast