polarofficial/polar-ble-sdk #607: Missing sleep data

viesti, 2025-05-19:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

**SDK version:**

6.1.0

**Customer app url**

Not yet available

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I wanted to ask that what could be reasons for missing sleep data from some nights?

For example, in the following output, I have few nights missing from `getSleepData()` results:

```
(-> (.getSleepData expo-polar
                   "2025-05-14T00:00:00Z"
                   "2025-05-19T23:59:00Z")
    (.then #(def sleep (->clj %)))
    (.catch #(def sleep %)))
#<js/Promise[~]>
(map :sleepEndTime sleep)
("2025-05-14T03:17:25Z" "2025-05-17T03:00:03Z" "2025-05-19T02:59:48Z")
```

I haven't yet ruled out bugs in the application that we are working on, we do call `stopSleepRecording()` to force a sleep report to be available early, could it be that we are calling `stopSleepRecording()` too early? The heuristic we have now is that we call `stopSleepRecording()` when date of last `stopSleepRecording()` (we store timestamp after calling `stopSleepRecording()`) differs from current date, but maybe we should add to that case, that we'd allow calling `stopSleepRecording()` only for example after 4:00 AM UTC, to give some room for firmware sleep recording to happen?

This then leads me to ask that is there a way to get from the firmware, the start time of sleep recording process, if there is such information present?

The SDK has `getSleepRecordingState()` and `observeSleepRecordingState()`, but `getSleepRecordingState()` returns only a boolean and for `observeSleepRecordingState()` it seems that the application should be running for it to get a notification on sleep recording state, which leads me to think that we should go implement wakeup of the app by Bluetooth events, which I'm not so sure of. To me it would seem easier if there would be a way to ask timestamp of last sleep recording state, which the application could then use to decide whether to tell the firmware to stop sleep recording.

Can there also be other reasons why sleep data would be missing for some days?

rkangast, 2025-05-20:
Hi @viesti 

We have identified these sleep deyecteion problems in Polar 360 firmware and there will be a fix released in firmware version 2.0 in June, exact release date has not been decided yet.

Currently it is not possible read any actual sleep data before the sleep ends and sleep data has been calculated in device. After you call stopSleepRecording api it will take some (tens of seconds) time for the device to calculate the sleep data. 

sleepStartTime field in tells the time when the sleep was detected and sleep recording was started, you can use that. You're correct with using  observeSleepRecordingState() api, it will wait for the device to answer to sleep state query until it receives answer. So you could wake up your app and use observeSleepRecordingState to query the device if the sleep measurement is ongoing or not.

viesti, 2025-05-20:
Nice, thank you for fast response! :)

closed 2025-07-04 by rkangast