polarofficial/polar-ble-sdk #609: Offline recording of HR and PPI

mikeread-dev, 2025-05-19:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ X] Platform is not relevant for this question

**SDK version:v6.2.1**

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X ] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

I occasionally hit the issue where the offline recordings (PPI and HR) stop on the device, often overnight.  This isn't due to low battery, low memory, after re-booting the device or after charing the device (which is an expected event) and it isn't something that happens regularly.

A couple of questions I have about this: 

1. Is it best practice to stop, delete , start offline recording with each sync from the device to the app (using Polar 360)?
2. Are there any other reasons, other than the ones I mentioned that would cause the recordings to stop?
3. Is there anything I can set in the device to automatically restart offline recordings if it stops without a manual call from the sdk?

rkangast, 2025-05-20:
Hi @mikeread-dev 

1. Offline recordings tend to grow to quite large files over time, and the memory in device is limited. Thus, your suggestion #1 would be really good. How long your recordings typically are (hours, days...)
2. Offline recordings should not stop on their own unless the device memory fills up. Which firmware version your device has?
3. No, there is not.

jimmyzumthurm, 2025-05-20:
For point 3, we have planned to add the Offline recording triggers as described here : https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/SdkOfflineRecordingExplained.md#advanced-usage

It means that the device would automatically restart these recordings for example after an unexpected reboot.

viesti, 2025-05-20:
Triggers sound neat!

Would there also be trigger to start offline recording after taking device out of charger? In our use case, we'd like recording to resume after charging, so it would be good for device to remember the recording state after charging (same might apply across reboot).

mikeread-dev, 2025-05-21:
@jimmyzumthurm   I actually replicated the issue where ppl and hr offline recording stopped last night.  Here are the relevant log items:

Battery and Memory have plenty of life and space:

```
[debug] | 8:35:05 920ms | Disk space: Total = 14 MB (15024128 bytes), Available = 10 MB (10088448 bytes)
[debug] | 8:34:54 521ms | Battery level: 79%
```

Tried to stop offline recordings during the morning sync, it was already in that state:

```[debug] | 8:34:06 138ms | Step 3: Stopping offline recording...
[warning] | 8:34:06 200ms | Warning: Error stopping PPI recording: Failed to stop offline recording: pmd cp command STOP_MEASUREMENT error: ERROR_ALREADY_IN_STATE: PlatformException(ERROR_STOPPING_RECORDING, pmd cp command STOP_MEASUREMENT error: ERROR_ALREADY_IN_STATE, null, null)
[debug] | 8:34:06 201ms | Continuing with sync process...
[warning] | 8:34:07 253ms | Warning: Error stopping HR recording: Failed to stop offline recording: pmd cp command STOP_MEASUREMENT error: ERROR_ALREADY_IN_STATE: PlatformException(ERROR_STOPPING_RECORDING, pmd cp command STOP_MEASUREMENT error: ERROR_ALREADY_IN_STATE, null, null)```

Offline records started at 8:30pm (PST):

```[debug] | 8:34:10 89ms | Entry 1: Type=PolarDataType.hr, Date=2025-05-21 04:30:59.000Z, Size=8985 bytes
[debug] | 8:34:10 90ms | Entry 2: Type=PolarDataType.ppi, Date=2025-05-21 04:30:57.000Z, Size=60927 bytes```

HR samples ran for around 2h30min: 

```[debug] | 8:34:14 123ms | Saved HR record with ID: 072027cb-b482-4573-ab45-be741f7f5df5, containing 147 minute averages (condensed from 8730 original samples)```

PPI samples likely ran the same amount of time:

```[debug] | 8:34:18 483ms | Successfully saved PPI record with 9900 samples to SDK
[debug] | 8:34:18 483ms | Processed 2 PPI and HR entries, 0 errors```

Assuming one per second for estimate: 165 mins

It looks like the recording stopped at 11pm to 11:30pm local time, the only thing I can think of for this time is that I took the band off to shower.  But I've don't this a lot and it normally just continues to record.

It stopped before my sleep record start time: 

```
[debug] | 8:34:18 388ms | Sleep Record:
[debug] | 8:34:18 389ms |   ID: 9b965d89-cb7b-4781-bb18-0cdf85631a43
[debug] | 8:34:18 389ms |   Start: 2025-05-21T08:10:08.000Z
[debug] | 8:34:18 389ms |   End: 2025-05-21T14:43:38.000Z
[debug] | 8:34:18 389ms |   Duration: 393 minutes
[debug] | 8:34:18 389ms |   Sleep: 355 minutes

```

I will add some more logging to our app to get the time of the last sample on the record, to see if can id the cause.  Any insight here would be gratefully received!