polarofficial/polar-ble-sdk #610: Device clock drift

viesti, 2025-05-21:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [ ] Platform is not relevant for this question

**SDK version:**

6.1.0

**Customer app url**

Not yet available

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

We noticed that the device clock can drift quite a lot, and wanted to ask that is this normal? We added now a check to the application where it adjusts device clock if it is off by 5 seconds compared to phone clock. I think this might be a bit too tight, but the worry is that when running offline recording constantly, drift in clock has an effect on data quality.

Are there known issues that could affect device clock to start drifting signifficantly?

viesti, 2025-05-21:
Good question, we have to set up more metric gathering, but I think I saw clock being off many hours, last clock sync probably done when I got the device few months ago.

mikeread-dev, 2025-05-21:
@viesti  I'm not sure if it's overkill, but we call `setLocalTime` each time we sync to the device.   It seems to keep it in sync pretty well.

viesti, 2025-05-22:
Nice to hear, since we are also doing the same :) It just occurred to me to ask if clock drifting is normal.

jimmyzumthurm, 2025-05-23:
@viesti I'll try to perform test over the week end to see what kind of drift I notice. It is normal to experience a small drift over time but I should be less than 1 second per 24 hours. The crystal oscillator used to clock the device should be well calibrated.

Otherwise setting the time regularly is a good practice to avoid any drifting error to accumulate over several days.