polarofficial/polar-ble-sdk #613: NO_SUCH_FILE_OR_DIRECTORY when using getSplitOfflineRecord

janbarari, 2025-05-22:
**Platform on which you observed the bug:** 
- [x] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
6.3.0

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Description of your observation
Dear Polar Team
I fetched the list of split offline records via `listSplitOfflineRecordings` and then try to get each one with `getSplitOfflineRecord` but everytime it throws `java.lang.Exception: NO_SUCH_FILE_OR_DIRECTORY` exception.

**How to Reproduce**
If you know how the bug may be reproduced please add the steps here
Offline recording

**Expected behavior**
Description of what you expected to happen. 
In what scenario this function works? What is the difference between `getSplitOfflineRecord` and `getOfflineRecord`?

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

jimmyzumthurm, 2025-05-22:
Hi @janbarari 

The goal is the split file features is that it allows partial sync of the recording. For example if you have 10 MB of ACC recording, it would be splitted into 20 x 500 KB split records that you can read and delete individually.

getOfflineRecord() would read everything at once -> 10 MB of ACC in 1 go
getSplitOfflineRecord() would read one of 20 sub recordings of the main ACC recording

janbarari, 2025-05-22:
Great that's exactly what we need. But I couldn't able to use it. Is there any specific limitation in terms of the time or size? For my test purposes I used 2 minutes record duration.

jimmyzumthurm, 2025-05-22:
There shouldn't be any specific limitation as long as there is data that has been flushed into the file.

We will investigate this internally and come back to you. 

Note that with the split file feature it is then possible to read a recording while it's ongoing (otherwise if using `getOfflineRecord() `API we advise to stop the recording first) , but you should not attempt to read the "open" sub-recording (maybe that would be the last element in the list returned by `listSplitOfflineRecordings() ` or the one that has not yet reach 500 KB in size).

janbarari, 2025-05-22:
So Imagine I have 250KB ACC data. In this case, Am I loosing the 250KB data even while the record is stopped?

jimmyzumthurm, 2025-05-22:
If you have only 250 KB of data, it would be advise to stop recording before reading it, but then it should work. The `listSplitOfflineRecordings()` should return 1 element and `getSplitOfflineRecord()` should then have same effect than `getOfflineRecord()`

janbarari, 2025-05-22:
I think there is an issue in `getSplitOfflineRecord()` because before I call it I will stop the recording and then trying to read the data. Maybe I need to add some delay between stop and get or increase my test duration to longer period since current tests are not even near 250KB data.

janbarari, 2025-05-22:
Adding delay between`stopOfflineRecording` and`getSplitOfflineRecord` didn't help(tried5, 10, 30 sec). 
While the record is stopped `getSplitOfflineRecord` function should work just like `getOfflineRecord`. I could do a temporary solution that use split option only the file is bigger than 500 KB. but I think it would be the same if i have like 600KB data which gonna be 2 part. part one will be received without issue but second one I think gonna throw the same exception.

rkangast, 2025-06-04:
@janbarari Fix has been released in SDK release 6.4.0.