polarofficial/polar-ble-sdk #614: Can't remove some offline recording files

l1va, 2025-05-22:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
6.3

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Some files can't be removed. 

**How to Reproduce**
getOfflineRecordings(), try to remove with 
NSLog("start offline recording removal")
let _: Void = try await api.removeOfflineRecord(deviceId, entry: offlineRecordingEntry).value
NSLog("offline recording removal completed")
logs say everything is ok, but then getOfflineRecordings returns these files again.                

**Expected behavior**
Files are removed or an exception is thrown.

**Screenshots and logs**

rkangast, 2025-05-22:
Hello @l1va  

I have created an internal ticket for this issue, FLOW-65173. We will look into this as soon as possible.

l1va, 2025-05-23:
@rkangast It seems temp recordings can't be removed, others work.

l1va, 2025-05-24:
@rkangast One more moment. It is possible to remove hr and ppi files when offline recording is in progress. And the status of the recordings is not changing - still recording (but I don't know where is it writing).  It seems like temperature's files are locked but hr and ppi's are not.
Looks related to bug #609

rkangast, 2025-05-26:
@l1va Removing the recording while the measurement is still in progress does not tell the device to stop the measurement. That is probably the reason the file appears back in the list. Why would you want to remove those files while recording is still in progress? Please, stop the measurement first 🙂 @jimmyzumthurm Any comments on why HR and PPI can recordings can be deleted while measurement is still in progress?

rkangast, 2025-05-28:
Hi @l1va 

I found out the reason that causes this problem for temperature offline recording files. Internal ticket FLOW-65370. This will be fix in the next SDK release.

jimmyzumthurm, 2025-05-28:
@l1va Deleting during measurement shouldn't be done as said in the SDK offline recording explained documentation, but it is allowed currently by the device. We have plans that the open recordings are not listed (like in Verity Sense) so that misuse of the feature is no longer possible.

Note that closed split files of ongoing recordings will be exposed for reading. The issue is only about deleting the file currently in use by the device.

l1va, 2025-06-09:
Thank you! Fixed.