polarofficial/polar-ble-sdk #615: Fields support, should they change?

l1va, 2025-05-25:
***Platform your question concerns:*** 
- [x] Platform is not relevant for this question

**SDK version:**
6.3

***Device:***
- [x] Polar 360

**Description:**
1. For HR I see the next fields: (hr: 53, ppgQuality: 0, correctedHr: 0, rrsMs: [], rrAvailable: false, contactStatus: false, contactStatusSupported: false). Only hr field is changing, are the others always empty? What are the conditions to see that they are changing too? 

2. For PPI I see the next fields: (timeStamp: 799757979042269103, hr: 61, ppInMs: 1279, ppErrorEstimate: 388, blockerBit: 0, skinContactStatus: 1, skinContactSupported: 1). I have not seen changes in skinContactStatus and skinContactSupported. Are they supposed to change over time? How to change them?   

3. ACC : I can't check the fields because of #601, but also good to know the list of really used fields.

pth-pe-gh, 2025-05-27:
Hi @l1va - for HR the corrected HR depends on 90 seconds of raw HR values collected first; so you should be seeing correctedHR values only after 90 seconds of continous HR measurement. contactStatusSupported false and rrAvailable means that corresponding values are not available. Let us get back on additional details later when available from Polar 360 team.

pth-pe-gh, 2025-05-30:
3. ACC: Here is documentation to the accelerometer data for [iOS](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Typealiases.html#/s:11PolarBleSdk0A7AccDataa) and [Android](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/model/PolarAccelerometerData.PolarAccelerometerDataSample.html)
Timestamp, x, y, and z-axis values should be all available in ACC recordings.

pth-pe-gh, 2025-06-09:
@I1va - more detail on corrected HR.  Currently correctedHR is available, starting from Polar 360 fw 1.4.38. You can use it when non-zero and fallback to HR field value whenever correctedHR value field is zero. Corrected HR, when available, should be more accurate HR value of the two.