polarofficial/polar-ble-sdk #617: Unable to Access Heart Rate, Battery, Temperature GATT Services and doFirstTimeUse Serialization via BLE GATT

gaurav36, 2025-05-26:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x ] Other
- [ ] Platform is not relevant for this question

**SDK version:** Latest

**Customer app url**
Link to app store / play store / your website

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x ] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

Hi Polar Team,

We are currently working on integrating the Polar device with our gateway system (via BLE GATT) as part of a project. We’ve been following the [polar-ble-sdk](https://github.com/polarofficial/polar-ble-sdk) and encountered a few challenges that we’d like to raise here for guidance.

### 1. **Pairing & Reconnection Issues (`pairingNull`)**
- We are using `pairingNull()` to connect the Polar device with our Linux-based gateway.
- Initially, the connection is successful, and data can be exchanged.
- However, after a period of inactivity, the device **stops advertising**, even though the **battery is charged**.
- The only way to re-enable advertising is to **physically connect the device to a charger and then disconnect it**.

**Question:**  
- 1. Is this expected behavior when using `pairingNull()`?  
- 2. Is there a recommended method to keep the device in advertising mode or re-enable it programmatically without physical interaction?

---

### 2. **Missing Documentation on GATT Service & Characteristics**
We were unable to get data like **Heart Rate, Temperature**, or **Battery Level** via BLE, despite trying the following:

- **Battery Level**
  - Service: `0000180f-0000-1000-8000-00805f9b34fb`
  - Characteristic: `00002a19-0000-1000-8000-00805f9b34fb`
- **Heart Rate**
  - Service: `0000180d-0000-1000-8000-00805f9b34fb`
  - Characteristic: `00002a37-0000-1000-8000-00805f9b34fb`
- **HTS (Health Thermometer)**
  - Service: `00001809-0000-1000-8000-00805f9b34fb`

We also attempted access via the **custom service** `fb005c80-02e7-f387-1cad-8acd2d8df0c8`, but did not find clear documentation.

**Request:**  
- 1. Could you share the correct and up-to-date list of **GATT service and characteristic UUIDs** supported by the Polar Verity Sense?  
- 2. Any relevant documentation would be appreciated.

---

### 3. **Unclear Encoding of `doFirstTimeUse` GATT Payload**
From the SDK, we identified that the first call expected after connection is `doFirstTimeUse()`, which requires writing a serialized `PolarFirstTimeUseConfig` object to:

- **Service ID**: `fb005c80-02e7-f387-1cad-8acd2d8df0c8`
- **Characteristic ID**: `fb005c81-02e7-f387-1cad-8acd2d8df0c8`

Sample config we are using:
```kotlin
PolarFirstTimeUseConfig(
    gender = PolarFirstTimeUseConfig.Gender.FEMALE,
    birthDate = birthDateCalendar.time,
    height = 165.5f,
    weight = 60.2f,
    maxHeartRate = 195,
    vo2Max = 45,
    restingHeartRate = 55,
    trainingBackground = 30,
    deviceTime = ZonedDateTime.now(ZoneOffset.UTC).format(DateTimeFormatter.ISO_DATE_TIME),
    typicalDay = PolarFirstTimeUseConfig.TypicalDay.MOSTLY_STANDING,
    sleepGoalMinutes = 450
)

- Could you recommend way to call this doFirstTimeUse using GATT service ? 

Thank you very much in advance

pth-pe-gh, 2025-05-27:
Disclaimer: Unfortunately, the SDK team is able to support only Android and iOS currently. The SDK includes source code and you can feel free to utilise / port the code in other platforms too but that is something we are not able to promise support for at the moment. 

1. i) Since pairingNull is not part of Polar BLE SDK, can not provide answer for this question. Also, Linux is not supported platform for Polar BLE SDK at this time, so that places limits on if and how you can use SDK on Linux.
1. ii) Polar 360 pairing: Perform a factory reset, this is supported by the SDK [on Android](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarBleApi.html#doFactoryReset(java.lang.String,java.lang.Boolean))
and [on iOS](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Protocols/PolarBleApi.html#/s:11PolarBleSdk0aB3ApiP14doFactoryReset_26preservePairingInformation7RxSwift17PrimitiveSequenceVyAF16CompletableTraitOs5NeverOGSS_SbtF)

The device can only be paired with one phone at a time. Once the device is paired with a phone, it will only be visible to that phone. Before the device can be paired with a new phone, it must be reset to factory settings to remove the previous pairing from it.

2. i) and ii) This information is not available at this time.

3. The recommended way is to use Polar BLE SDK doFirstTimeUse() method on the currently supported platforms. See API documentation for [Android](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarBleApi.html#doFirstTimeUse(java.lang.String,com.polar.sdk.api.model.PolarFirstTimeUseConfig)) and [iOS](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Protocols/PolarBleApi.html#/s:11PolarBleSdk0aB3ApiP14doFirstTimeUse_9ftuConfig7RxSwift17PrimitiveSequenceVyAF16CompletableTraitOs5NeverOGSS_AA0afghJ0VtF)

Should you require further assistance with Linux based integrations, the best way to proceed would be to first get in touch with our Polar for Business contacts. A form to initiate conversation how Polar may be of help in your case is available at https://www.polar.com/en/business

gaurav36, 2025-05-27:
> Disclaimer: Unfortunately, the SDK team is able to support only Android and iOS currently. The SDK includes source code and you can feel free to utilise / port the code in other platforms too but that is something we are not able to promise support for at the moment.
> 
> 1. i) Since pairingNull is not part of Polar BLE SDK, can not provide answer for this question. Also, Linux is not supported platform for Polar BLE SDK at this time, so that places limits on if and how you can use SDK on Linux.
> 2. ii) Polar 360 pairing: Perform a factory reset, this is supported by the SDK [on Android](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarBleApi.html#doFactoryReset(java.lang.String,java.lang.Boolean))
>    and [on iOS](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Protocols/PolarBleApi.html#/s:11PolarBleSdk0aB3ApiP14doFactoryReset_26preservePairingInformation7RxSwift17PrimitiveSequenceVyAF16CompletableTraitOs5NeverOGSS_SbtF)
> 
> The device can only be paired with one phone at a time. Once the device is paired with a phone, it will only be visible to that phone. Before the device can be paired with a new phone, it must be reset to factory settings to remove the previous pairing from it.
> 
> 2. i) and ii) This information is not available at this time.
> 3. The recommended way is to use Polar BLE SDK doFirstTimeUse() method on the currently supported platforms. See API documentation for [Android](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarBleApi.html#doFirstTimeUse(java.lang.String,com.polar.sdk.api.model.PolarFirstTimeUseConfig)) and [iOS](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Protocols/PolarBleApi.html#/s:11PolarBleSdk0aB3ApiP14doFirstTimeUse_9ftuConfig7RxSwift17PrimitiveSequenceVyAF16CompletableTraitOs5NeverOGSS_AA0afghJ0VtF)
> 
> [@gaurav36](https://github.com/gaurav36) Should you require further assistance with Linux based integrations, the best way to proceed would be to first get in touch with our Polar for Business contacts. A form to initiate conversation how Polar may be of help in your case is available at https://www.polar.com/en/business

Hi @pth-pe-gh , Many thanks for your reply. 

We are also using pairingJustWork and we are facing the same issue. Our use case also not tied specific to Linux or something, we are just using BLE GATT service that SDK provided by you also using. 

With BLE GATT service it should work as well as its working with SDK because underlying GATT service is the same. 

For the 2nd point, you mentioned this information is not available, but i can clearly see it on code base. for example: 
-  https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/Android/android-communications/library/src/main/java/com/polar/androidcommunications/api/ble/model/gatt/client/BleBattClient.kt#L27 
- https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/Android/android-communications/library/src/main/java/com/polar/androidcommunications/api/ble/model/gatt/client/BleHrClient.kt#L31
- https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/Android/android-communications/library/src/main/java/com/polar/androidcommunications/api/ble/model/gatt/client/pmd/BlePMDClient.kt#L656

These are the same service we are trying. and during service discovery that is very standard to Bluetooth communication we are getting similar result for above mentioned service and characteristic ID 

for the third point FTU (first time use), we tried following:

```bash
{
    "technology": "ble",
    "id": "b063a447-c3c2-40cd-9a2b-0050aa6b866b",
    "value": "AYCdniUAgCVDzcxwQsMtNx4CwgGFgTRo",
    "ble": {
        "serviceID": "fb005c80-02e7-f387-1cad-8acd2d8df0c8",
        "characteristicID": "fb005c81-02e7-f387-1cad-8acd2d8df0c8"
    },
    "controlApp": "controlApplication"
}
```

Here I used `  "value": "AYCdniUAgCVDzcxwQsMtNx4CwgGFgTRo", ` base64 encoding for following parameter:

```bash
val ftuConfig = try {
    PolarFirstTimeUseConfig(
        gender = PolarFirstTimeUseConfig.Gender.FEMALE,
        birthDate = birthDateCalendar.time,
        height = 165.5f, // cm
        weight = 60.2f,  // kg
        maxHeartRate = 195, // bpm
        vo2Max = 45, // ml/kg/min
        restingHeartRate = 55, // bpm
        trainingBackground = 30, // valid values [10, 20, 30, 40, 50, 60]
        deviceTime = ZonedDateTime.now(ZoneOffset.UTC).format(DateTimeFormatter.ISO_DATE_TIME), // Current UTC time in ISO 8601
        typicalDay = PolarFirstTimeUseConfig.TypicalDay.MOSTLY_STANDING,
        sleepGoalMinutes = 450 // 7.5 hours
    )
```
And above API call was successful but LED color did not come to after it come successfully by executing FTU call. 

We are also in connect with Polar Business contact, but it does not help till now.

gaurav36, 2025-06-02:
@pth-pe-gh thanks for your answer. We are using polar360 and Verity sense is not scope at this time.

for FTU function  i see from SDK this doFirstTimeUse code https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/BDBleApiImpl.kt#L2225 

based on this we used following payload (together with serviceID and characteristicID):

```bash
    "value": "AQHXCgMFD1cGXAJcAkPjLTcGHDIwMjQtMDUtMjBUMTI6MzQ6NTYuNzg5WgACCsI=",
    "ble": {
        "serviceID": "fb005c80-02e7-f387-1cad-8acd2d8df0c8",
        "characteristicID": "fb005c81-02e7-f387-1cad-8acd2d8df0c8"
    },
```
In above value that is basically FTU configuration for following is base64 encoding

```bash
gender = PolarFirstTimeUseConfig.Gender.FEMALE,
        birthDate = birthDateCalendar.time,
        height = 165.5f, // cm
        weight = 60.2f,  // kg
        maxHeartRate = 195, // bpm
        vo2Max = 45, // ml/kg/min
        restingHeartRate = 55, // bpm
        trainingBackground = 30, // valid values [10, 20, 30, 40, 50, 60]
        deviceTime = ZonedDateTime.now(ZoneOffset.UTC).format(DateTimeFormatter.ISO_DATE_TIME), // Current UTC time in ISO 8601
        typicalDay = PolarFirstTimeUseConfig.TypicalDay.MOSTLY_STANDING,
        sleepGoalMinutes = 450 // 7.5 hours
```

Executing above is successful but LED color still in waiting for firstTimeUse state (all red color still), to make complete execution of FTU function, do we need to execute any other call with same serviceID or characteristicID with different payload like setting local time etc...?

I can see sample application here  https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/polar-sensor-data-collector but not much information

gaurav36, 2025-06-02:
@pth-pe-gh Thanks for this information. I found GATT ServiceID and Characteristic ID form source code. but my original question 3rd part is not working yet, that is executing doFirstTimeUse  , via GATT serviceID and CharesticID along with payload formation. I tried to use base64 encoding for payload and it executed successfully but device did not get initialized. How we used it, mentioned in original question and https://github.com/polarofficial/polar-ble-sdk/issues/617#issuecomment-2929284497

palmqvisti, 2025-06-10:
@gaurav36, have you taken a look at our more comprehensive example app PSDC? 
https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/polar-sensor-data-collector
It includes example code of all api features. 

Like mentioned earlier in this thread we only provide support for iOS and Android SDK.