polarofficial/polar-ble-sdk #622: Calling getOfflineRecordingTriggerSetup results into "Invalid op code" error

viesti, 2025-06-01:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
6.3.0

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**

Calling `getOfflineRecordingTriggerSetup()` throws and error:

```
gattAttributeError(errorCode: 1, errorDescription: "Invalid op code")
```

**How to Reproduce**

Connect to device and call `getOfflineRecordingTriggerSetup()`. I didn't specify a trigger setup before the call, not sure if this is related.

**Expected behavior**

`getOfflineRecordingTriggerSetup()` should not throw, even if no trigger setup done.

**Screenshots and logs**

Here is a Console log from the application:

```
default	16:48:13.405745+0300	PolarBleApiLogger: [BLE] Get offline recording trigger setup. Device: E4937920
default	16:48:13.430995+0300	PolarBleApiLogger: PMD command:  HEX F0 07 00 01 01
default	16:48:13.431058+0300	Error while getOfflineRecordingTriggerSetup gattAttributeError(errorCode: 1, errorDescription: "Invalid op code")
```

viesti, 2025-06-01:
Seems that calling `setOfflineRecordingTrigger()` results in the same error, since it internally does a read via the same `getOfflineRecordingTriggerStatus()` as `getOfflineRecordingTriggerSetup()` does.

jimmyzumthurm, 2025-06-02:
@viesti This offline trigger feature is not available yet but will be soon in the next coming release.

viesti, 2025-06-02:
Ah, thank you! :)