polarofficial/polar-ble-sdk #623: Why bluetooth pairing via system bluetooth prevents app connecting

viesti, 2025-06-04:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

**SDK version:**
6.3.0

**Customer app url**

Not yet available

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I might jump the gun on asking this, but please bear with me :)

The Polar 360 instructions say to pair with the app, not via system settings, and if done via system bluetooth settings, pairing needs to be undone on both sides (via factory reset on device side).

[image]

What is the reason for this? Is it a SDK side thing, or is it a firmware side thing? Most consumer hardware allows pairing via Bluetooth settings, and then use by apps, so it might come across odd for consumers used to that style of pairing.

This might have been explained in other material, so I might have skimmed over this, so I apologize for that.

I started to think if this is application specific security measure (like prevent other applications having access to health measurements provided by the hardware), but then started to think that this kind of thing might need a kind of encryption key setup where the application should exchange and persist securely, a key with which the firmware encrypts data, which kind of feels like and end-to-end application level security thing, not a bluetooth security thing.

Anyways, just felt that should ask, thank you! :)

samulimaa, 2025-06-04:
Hi,

Devices are designed to pair through the app so that the connection is initialized properly. Probably bypassing app pairing would cause issues, for example custom GATT services may not be discovered.
So pairing from the phone Bluetooth settings isn’t currently supported.

viesti, 2025-06-24:
A friendly bump here, I think when launching our app, we'll probably get support tickets around this subject, since people will probably try pairing via system Bluetooth first before reading the manual.

> for example custom GATT services may not be discovered

Is there any way to go around this?