polarofficial/polar-ble-sdk #625: Consider a "flush" API for offline recording

viesti, 2025-06-04:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

**SDK version:**
6.3.0

**Customer app url**

Not yet available

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

The procedure for using offline recording seems the following:

1. Start offline recording and let it run
2. To download data:
   1. Stop offline recording
   2. List files
   3. Download files
   4. Resume offline recording 💥

It can happen that the 2.4 step of "resume offline recording" fails, for example if device goes out of radio range (or the app is killed). This can then lead to data loss, since the user of the API was not able to restart offline recording.

This led me to think of a different procedure:

1. Start offline recording
2. To download data
    1. Call "flush", which would flush current buffer and start a new file, and continue offline recording
    2. List files, excluding newly created "active" file. Or, optionally, include in file metadata, if file is being actively written to
    3. Download files

In this operation, it would not be possible to accidentally miss the resuming of offline recording by failing to send start offline recording command after download (or after file listing).

I think this operation procedure would resemble the 247 sample taking, in that I think the 247 sample taking is not something that user can "accidentally" stop.

mikeread-dev, 2025-06-04:
+1 for this, I constantly check for running offline recordings.  This will be alleviated a bit once we can set the auto start triggers, but it would be really nice to have the peace of mind that offline recordings are running after a charge, sync process.   

The ability to just get ppi / hr recorded data for a date/time range would be my ideal, but i'm guessing that would be a bit of a battery killer and too specific for some use cases.

viesti, 2025-06-27:
Thanks for the response! :)

> In Verity Sense this problem isn't there as the file that just got created is not exposed to file listing until it is closed.

Ah right, didn't realise that firmware can also choose to not list active file. Thinking that the SDK knowing which file is "active" can be a useful thing too.

Another thought is that the 24/7 data gathering is a bit similar, and that it is more robust in the sense that, if I understand correctly, it is a continuous background operation, not something that the user can interrupt, which can then lead to problems. 

But one probably doesn't want 24/7 offline recording, but a kind of controlled offline recording that is also robust at synchronization points.

viesti, 2025-07-04:
We have an idea:

1. List files => snapshot of file list
2. Stop offline recording
3. Resume offline recording
4. Download snapshotted files

Idea here being that listing files gives a snapshot, that can also contain "unready" files, meaning those that haven't yet had their corresponding volatile memory buffer flushed to non-volatile memory. But, this isn't an issue, since calling `stop` after `list`, would then flush those unready files, and we would have a snapshot list of files, that should be safe to download.

The upside is that we don't need to wait the duration of the download before starting recording again, which makes for less error margin.

This stop+start could be a "flush" operation even. An operation that the firmware does, which blocks for duration of the operation.

What do you think about this idea?